micropython/micropython #7738: PYBD SF6W WiFi Issues - [CYW43] HT not ready

bensherlock, 2021-09-01:
So far seven of our PYBD SF6W devices have started to have problems with the wifi module. When setting the wifi module to be active(True) an error message is printed to the console and the wifi is unable to be used. I have 7 devices that now show this behaviour, that in the past have worked fine connecting to a wifi access point and downloading files over http. 

I have run the same set of commands using rshell/repl on a number of versions of the firmware, including the latest unstable release. I found it produced extra output statements on the v1.12 firmware that might be more of a clue. 

This issue is related to the [forum post](https://forum.micropython.org/viewtopic.php?f=20&t=11061).

As well as the "[CYW43] HT not ready", messages I have also occasionally seen  "[CYW43] F2 not ready".

```
Entering REPL. Use Control-X to exit.
>
MicroPython v1.12 on 2019-12-20; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>>
>>> import network
>>> sta_if = network.WLAN(network.STA_IF)
>>> sta_if.config(trace=1)
>>> sta_if.active()
False
>>> sta_if.active(True)
[CYW43] HT not ready
[CYW43] HT not ready
[CYW43] HT not ready
[CYW43] send_ethernet failed: -5
[CYW43] HT not ready
[CYW43] send_ethernet failed: -5
>>>
```

```
Entering REPL. Use Control-X to exit.
>
MicroPython v1.16-259-g86371781e on 2021-08-31; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>>
>>> import network
>>> sta_if = network.WLAN(network.STA_IF)
>>> sta_if.config(trace=1)
>>> sta_if.active()
False
>>> sta_if.active(True)
[CYW43] HT not ready
>>>
```

Is it possible that the firmware on the SoC WiFi chip has been corrupted? Is there a way to reflash this chip?

dpgeorge, 2021-09-01:
> Is it possible that the firmware on the SoC WiFi chip has been corrupted?

This is unlikely.  The CYW4343 firmware is downloaded to the chip each time it is powered on (during .active(True) the first time after a power cycle; and a deinit() will power it down).

Does Bluetooth work?  Try:
```
import bluetooth
ble = bluetooth.BLE()
ble.active(1)
ble.irq(lambda *x: print(*x))
ble.gap_scan(5000, 100, 100)
```
You should see lots of lines printed, scanning for BLE devices.

bensherlock, 2021-09-01:
OK, thanks for the info. So assuming comms with the SoC is ok then it should get the latest firmware. 

Trying your bluetooth code with firmware version:
`MicroPython v1.16-259-g86371781e on 2021-08-31; PYBD-SF6W with STM32F767IIK`

The device never returns from the `ble.active(1)` command. Using ctrl-c and entering the subsequent commands:

```
Entering REPL. Use Control-X to exit.
>
MicroPython v1.16-259-g86371781e on 2021-08-31; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>>
>>>
>>>
>>> import bluetooth
>>> ble = bluetooth.BLE()
>>> ble.active(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt:
>>> ble.irq(lambda *x: print(*x))
>>> ble.gap_scan(5000, 100, 100)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
>>>

```

dpgeorge, 2021-09-03:
> So assuming comms with the SoC is ok then it should get the latest firmware.

Yes, that's correct.

> Trying your bluetooth code with firmware version:

So you are reflashing the PYBD-SF6 with the latest firmware?  How do you deploy this firmware (DFU, SWD)?  One possibility is that the WiFi/BT firmware (which is bundled in the .dfu file) is corrupt and hence crashing the WiFi/BT SoC.  And reflashing the PYBD-SF6 should fix that if that is the issue.

> The device never returns from the `ble.active(1)` command.

That is very interesting.  The WiFi and BT parts of the CYW4343 SoC are pretty much independent so if they are both not working (not starting up) then it does point to invalid firmware being downloaded to them.

> If the firmware for the SoC is provided each time active(1) is called, is this written to flash on the SoC or just executed from RAM? What is the likely limit on how many times this could be written before the flash degrades?

It's written to RAM on the CYW4343 SoC.  Both the WiFi and BT parts of the SoC have firmware written to them, independently, and both to RAM.  So there shouldn't be any limit on the number of times they can be activated/power cycled.

> I had 10 devices running for several months and all 10 of them are now showing the same problem.

Did they all fail after running for about the same time?  Were there any other factors that would correlate with them failing (eg new firmware version put on all of them, some electrical activity like a storm, etc)?

> However, as the batteries ran all the way down to flat

How are the PYBD being powered?  There are constraints on the power supply voltage needed by the CYW4343 SoC.  Is it possible the power failed during a firmware/software update?

bensherlock, 2021-09-07:
> So you are reflashing the PYBD-SF6 with the latest firmware? How do you deploy this firmware (DFU, SWD)? One possibility is that the WiFi/BT firmware (which is bundled in the .dfu file) is corrupt [...]

I've been using the dfu-util to flash the boards. Throughout the testing I've flashed the boards under test with various versions - 1.12 through to 1.16 as well as one of the unstable releases (31st August version). I would assume that this would overwrite any corrupted binary that is pushed to the SoC? Is there any way to check this?

> It's written to RAM on the CYW4343 SoC. Both the WiFi and BT parts of the SoC have firmware written to them, independently, and both to RAM. So there shouldn't be any limit on the number of times [...]

OK, that's good to be able to rule out flash write limits then.

> Did they all fail after running for about the same time? Were there any other factors that would correlate with them failing (eg new firmware version put on all of them, some electrical activity [...]

They were all indoors, in the same room, and powered with batteries that were fitted around similar times. They were left there for quite a while so when finally going to prepare them and swap out the batteries this new issue became apparent. There aren't any obvious sources of large electrical discharge etc that could have affected them.

> > However, as the batteries ran all the way down to flat
> 
> How are the PYBD being powered? There are constraints on the power supply voltage needed by the CYW4343 SoC. Is it possible the power failed during a firmware/software update?

Power to the PYBD comes from 4of alkaline C cells in series which is then linear regulated with a LT1763-ADJ set to output of 3.6V. This is then provided at VIN on the PYBD and further regulated down to 3.3V with the onboard regulator. 

As the first thing the PYBD software did was to activate the wifi SoC (with active(1)), it's possible that the system browned out during the time the drivers were pushing the firmware to the SoC or at least when the SoC was attempting to connect.

If it is helpful for your diagnostics at the hardware level we could post one of the PYBDs back to you?

kevinjwalters, 2023-08-31:
I can make these error messages on a Pico W from a C++ program which is setting up continuous reads via DMA from GPIO26 ADC then trying to run Bluetooth stuff: [Raspberry Pi Forums: DMA ADC vs Bluetooth/Wi-Fi on Pico W?](https://forums.raspberrypi.com/viewtopic.php?t=355701)