micropython/micropython #7753: ESP32 port need a "make clean" to freeze custom py module

Patoche5150, 2021-09-07:
Hi,

I work with the new cmake toolchain and the last micropython version. Everithing is ok, I am able to build the ESP32 port. 
However with this latest version of micropython, as soon as I add a new python module in the module directory or if I update one, the "make" command does not detect the new files and I have to do a `make clean` before each `make`.
If I modify a .C file, it is detected and the build is ok.
I checked the manifest.py file in the boards directory and the line `freeze("$(PORT_DIR)/modules")` is present.
My py modules are in this directory.

Previously, I worked with the 1.13 version and I didn't have this behaviour, each py modification was detectet with the make command.
Maybe I miss to modify a configuration file or something but I didn't find a solution for that.

Thanks

chriscmorgan, 2021-09-27:
I have the same problem, unable to build esp32 generic or generic_C3 with the latest micropython version.
If I go back to commit "7c54b64" it builds fine.

pillo79, 2021-11-05:
I have the same problem as well, easily repeatable on v1.17 (`7c54b64`) or with current HEAD: just do a full build, then touch a file in the MPY folder, and observe no recompilation at all.

I am pretty sure the issue lies between cmake and ninja regarding forced recompilation.
The code generated by cmake for the `MICROPY_FORCE_BUILD` target in `build.ninja` is:

```
  build esp-idf/main/MICROPY_FORCE_BUILD: CUSTOM_COMMAND || esp-idf/app_trace/libapp_trace.a esp-idf/app_update/libapp_update.a esp-idf/bootloader_support/libbootloader_support.a esp-idf/bt/libbt.a esp-idf/console/libconsole.a esp-idf/cxx
    COMMAND = cd /home/ports/esp32/build-GENERIC/esp-idf/main && echo -n
    restat = 1     
```

According to the [Ninja build system docs](https://ninja-build.org/manual.html#_the_literal_phony_literal_rule) on the `phony` target: 

> `phony` can also be used to create dummy targets for files which may not exist at build time. If a phony build statement is written without any dependencies, the target will be considered out of date if it does not exist. **Without a phony build statement, Ninja will report an error if the file does not exist and is required by the build.**

(emphasis mine).
In fact, if I run `ninja -d explain -v all` in the build directory I can see that something goes horribly wrong: 
```
ninja explain: output esp-idf/mbedtls/custom_bundle of phony edge with no inputs doesn't exist
ninja explain: output esp-idf/main/MICROPY_FORCE_BUILD doesn't exist
ninja explain: esp-idf/main/MICROPY_FORCE_BUILD is dirty
ninja explain: esp-idf/main/MICROPY_FORCE_BUILD is dirty
ninja explain: genhdr/mpversion.h is dirty
ninja explain: genhdr/moduledefs.h is dirty
ninja explain: genhdr/qstr.i.last is dirty
ninja explain: genhdr/qstr.split is dirty
ninja explain: genhdr/qstrdefs.collected.h is dirty
ninja explain: genhdr/qstrdefs.preprocessed.h is dirty
ninja explain: genhdr/qstrdefs.generated.h is dirty
ninja explain: genhdr/qstrdefs.generated.h is dirty
... snip lots of other dirty things ...
ninja explain: esp-idf/main/all is dirty
ninja explain: esp-idf/all is dirty
[1/209] cd /home/ports/esp32/build-GENERIC/esp-idf/main && echo -n
[2/4] cd /home/ports/esp32/build-GENERIC/bootloader && /opt/esp/tools/cmake/3.16.4/bin/cmake --build .
ninja: no work to do.
```

That 1/209 is the "rebuild esp-idf/main/MICROPY_FORCE_BUILD" step... that does not produce an actual file named that way. All processing for dependent targets (basically everything!) is thus dropped and the count is quicky [2/4].

I have found that if I edit the `build.ninja` file and replace the generated rule with a simple
```
build esp-idf/main/MICROPY_FORCE_BUILD: phony
``` 
... everything works as expected. 
However, I am unable to convince cmake to output that... 😞 The closest I got was with this change:

```
diff --git a/py/mkrules.cmake b/py/mkrules.cmake
index 9b5381975..e4783ee8b 100644
--- a/py/mkrules.cmake
+++ b/py/mkrules.cmake
@@ -36,13 +36,9 @@ target_sources(${MICROPY_TARGET} PRIVATE
     ${MICROPY_QSTRDEFS_GENERATED}
 )
 
-# Command to force the build of another command
+# Target to force the build of another target
 
-add_custom_command(
-    OUTPUT MICROPY_FORCE_BUILD
-    COMMENT ""
-    COMMAND echo -n
-)
+add_custom_target( MICROPY_FORCE_BUILD )
```
... but still does not work. Any help from cmake or ninja gurus is appreciated!

pillo79, 2021-11-05:
Dirty, ugly, ashamed-to-post "fix" for those so annoyed that they'll do anything to solve the immediate issue: create that file with a date properly in the future, so all targets that depend on that always get rebuilt.

```
diff --git a/py/mkrules.cmake b/py/mkrules.cmake
index 9b5381975..8e40302bc 100644
--- a/py/mkrules.cmake
+++ b/py/mkrules.cmake
@@ -41,7 +41,7 @@ target_sources(${MICROPY_TARGET} PRIVATE
 add_custom_command(
     OUTPUT MICROPY_FORCE_BUILD
     COMMENT ""
-    COMMAND echo -n
+    COMMAND touch -t 222202020202 MICROPY_FORCE_BUILD
 )
 
 # Generate mpversion.h
```
Still looking forward to some help for the proper Fix though! ;-)

glenn20, 2022-12-03:
Yes - this is annoying. The Cmake files do not declare that ${MICROPY_FROZEN_CONTENT} has a dependency on the mpy-cross output files (build_XXX/frozen_mp/**/*.mpy). Cmake is a mystery to me, so I just use:

```bash
rm build_GENERIC/frozen_content.c
```
then my make command line.

If you remove frozen_content.c it will re-cross-compile only those modules which have been changed.

PR #10691 (2023-02-09): py/mkrules.cmake: Force build mpversion.h and frozen_content.c.
This ensures that all builds unconditionally run makeversionhdr.py and makemanifest.py to generate mpversion.h and frozen_content.c respectively. This now matches the Makefile behavior, and in particular this fixes the issue on ESP32 builds that changes in code-to-be-frozen will cause the build to update. Both these already tools know not to touch their output if there is no change.

Fixes #7753 (and several reports in discussions)

_This work was funded through GitHub Sponsors._
closed 2023-02-09 by dpgeorge