micropython/micropython #7772: Hardware I2C fail on ESP32_C3 - timing value error

haybarry, 2021-09-11:
Hardware I2C fails on ESP32_C3 with i2c timing value error.

i2c=I2C(0, scl=Pin(5), sda=Pin(4))    

fails with error 
E (87717) i2c: i2c_set_timeout(783): i2c timing value error.

Hardware:  Lilygo T-OI Plus V1.0  ESP32-C3 (revision 3)

Micropython:
Version v1.17-20-g0a5107372  
Compiled on RPi 4 with idf4.3

Same MicroPython compiled for ESP32 or ESP32_S2 successfully instantiates i2c 

--Barry

fivdi, 2022-01-29:
This error also occurs with MicroPython v1.18 on a [LILYGO T-01C3](https://github.com/Xinyuan-LilyGO/T-01C3). Here is an example of the error occuring in the REPL:
```
MPY: soft reboot
MicroPython v1.18 on 2022-01-17; ESP32C3 module with ESP32C3
Type "help()" for more information.
>>> from machine import Pin, I2C
>>> i2c = I2C(0, scl=Pin(8), sda=Pin(2), freq=800000)
E (45966) i2c: i2c_set_timeout(817): i2c timing value error
>>> i2c.scan()
[24]
>>> 
```
As can be seen, the error message `E (45966) i2c: i2c_set_timeout(817): i2c timing value error` is printed when creating `i2c`. However, I2C methods like `scan`, `writeto` and `readfrom` still function as expected.

The following call to `i2c_set_timeout` causes the error:

https://github.com/micropython/micropython/blob/da4b38e7562dfa451917f9d7f344a7f26de8c7bd/ports/esp32/machine_i2c.c#L74

The function prototype for `i2c_set_timeout` is identical for all the different ESP32 cores (ESP32, ESP32C3, ...) and is defined [like this](https://github.com/espressif/esp-idf/blob/v4.3.1/components/driver/include/driver/i2c.h#L480):
```c
esp_err_t i2c_set_timeout(i2c_port_t i2c_num, int timeout);
```

However, the semantics of the `timeout` parameter varies from core type to core type.

On an ESP32, `timeout` represents the number of AHB clock cycles that need to occur before the timeout expires. The `timeout` value passed is stored in the lower 20 bits of register `I2C_TO_REG`.

On an ESP32C3, the number of AHB clock cycles that need to occur before the timeout expires is 2 ** timeout. The `timeout` value passed is stored in the lower 5 bits of register `I2C_TO_REG`.

sugarpines, 2022-04-19:
This issue caused me issues because the clock was running too fast for my target. I found setting freq=50000 configured I2C with a frequency of 100000 which was a satisfactory work around until addressed.

RetiredWizard, 2022-05-17:
I've encountered this error on the ESP32-S3-DevKitC-1 as well:

    MicroPython v1.18 on 2022-05-17; ESP32S3 module (spiram) with ESP32S3
    Type "help()" for more information.
    >>>
    >>> from machine import Pin, I2C
    >>> i2c = I2C(0,scl=Pin(7),sda=Pin(6))
    E (1405330) i2c: i2c_set_timeout(817): i2c timing value error
    >>> i2c.scan()
    [38, 39]
    >>>

dhalbert, 2022-05-17:
There are various reported issues with ESP32-S3 and ESP32-C3 in https://github.com/espressif/[esp-idf](https://github.com/espressif/esp-idf). Some have been fixed, but only just recently, and some fixes are not yet in a release. For instance, we (CircuitPython) are seeing problems with at least one sensor that's using clock stretching on S3: https://github.com/espressif/esp-idf/issues/8894.

dpgeorge, 2022-05-18:
Until this is fixed, you can use `machine.SoftI2C` instead of the hardware I2C implementation.

puppet13th, 2022-07-07:
> Until this is fixed, you can use `machine.SoftI2C` instead of the hardware I2C implementation.

I have the opposite experience.
on fresh boot, softI2C does not detect any i2c peripheral when i2c.scan() issued
after hardware I2C initialize,softI2C can detect i2c peripheral
example code:
```
from machine import Pin, SoftI2C,I2C
scl = 18
sda = 19

i2c = SoftI2C(scl=Pin(scl),sda=Pin(sda),freq=20_000)
print(f"softI2C : {i2c.scan()}")

i2c = I2C(0,scl=Pin(scl),sda=Pin(sda),freq=20_000)
print(f"I2C(0) : {i2c.scan()}")
```
output:
```
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0xe3c
load:0x403ce000,len:0x6f4
load:0x403d0000,len:0x28d8
entry 0x403ce000
MicroPython v1.19.1 on 2022-07-05; ESP32C3 module with ESP32C3
Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
softI2C : []
I2C(0) : [80]
>>> %Run -c $EDITOR_CONTENT
softI2C : [80]
I2C(0) : [80]
```

mgsb, 2022-09-26:
I believe I have figured out the issue. See this PR: https://github.com/micropython/micropython/pull/9434

dpgeorge, 2022-10-28:
Should be fixed by 12f99481518b0ebcb14f00b2323865a845c2a4f1

closed 2022-10-28 by dpgeorge

alprshn, 2023-10-11:
bro try it. I solved problem
SSD1306Wire display(0x3c, I2C_SDA, I2C_SCL, GEOMETRY_128_64, I2C_ONE, -1);
You can look this link https://github.com/ThingPulse/esp8266-oled-ssd1306