micropython/micropython #7776: Error when copying file to ESP32 with mpremote

Friday8229, 2021-09-12:
The board is running micropython env. I want to upload a CSV file but there's an error. Also I can't upload any .py file.

```
friday@friday-VirtualBox:~/esp-idf/micropython/ports/esp32$ mpremote cp test.py :
cp test.py :test.py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV

```

when using pyboard.py :

```
friday@friday-VirtualBox:~/esp-idf/micropython/ports/esp32$ ./pyboard.py --device /dev/ttyUSB0 -f cp A00068.csv :A00068.csv
cp A00068.csv :A00068.csv
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```

robert-hh, 2021-09-12:
For esp32, the device has to be e.g. /dev/ttyUSB0
for Pyboard, it has to be /dev/ttyACM0
It may as well be USB1, USB2, .. or ACM1, ACM2, ..., but ESP32 can be found under the USBx devices, PyBoard or RP2 under the ACMx devices. The default device of mpremote is /dev/ttyACM0. The are shortcuts as:
a0, a1, a2, u0, u1, u2.
Usage:
mpremote u0 cp test.py :

Friday8229, 2021-09-12:
I can access the device from virtual box, I used it to flash the micropython env. I'm getting the same error after trying the command you suggested 

for reference : 

```
friday@friday-VirtualBox:~/esp-idf/micropython/ports/esp32$ mpremote
Connected to MicroPython at /dev/ttyUSB0
Use Ctrl-] to exit this shell

>>> 

```

Error:

```
friday@friday-VirtualBox:~/esp-idf/micropython/ports/esp32$ mpremote u0 cp test.py :
cp test.py :test.py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV

```

robert-hh, 2021-09-12:
Do you have a second window open with a connection to the ESP32?
What is the result of e.g.:
mpremote u0 ls

Friday8229, 2021-09-12:
No, i don't have a second window.

```
friday@friday-VirtualBox:~/esp-idf/micropython/ports/esp32$ mpremote u0
Connected to MicroPython at /dev/ttyUSB0
Use Ctrl-] to exit this shell

>>> print ("HI!")
HI!
>>> 

```

robert-hh, 2021-09-12:
So the mpremote connection is fine. When at the repl promt, what is the result of:

import uos
uos.listdir("")

Friday8229, 2021-09-12:
Here are the results 

```
friday@friday-VirtualBox:~/esp-idf/micropython/ports/esp32$ mpremote u0
Connected to MicroPython at /dev/ttyUSB0
Use Ctrl-] to exit this shell

>>> import uos
>>> uos.listdir("")
[]
>>> 
```

robert-hh, 2021-09-12:
Since you do not get an error, there is a file system, and it is empty. Strange, that cp does not work. Can you try to create a file at the repl prompt, like:

import uos
uos.listdir("")
uos.getcwd()

f=open("test.py", "w")
f.write("print('Hi!')\n")
f.close()

uos.listdir("")

Friday8229, 2021-09-12:
I get an error while opening the file

```
friday@friday-VirtualBox:~/esp-idf/micropython/ports/esp32$ mpremote u0
Connected to MicroPython at /dev/ttyUSB0
Use Ctrl-] to exit this shell

>>> import uos
>>> uos.listdir("")
[]
>>> uos.getcwd()
'/'
>>> f=open("test.py", "w")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV

```

To know what i did checkout this issue #7705, In a nutshell i have a custom partition table and c++ code running on the board.

The partition table:

```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap,,,,
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 4M,
# created the spiffs partition for the storage of cvs file
storage,  data, spiffs,  ,        1M, 
```

robert-hh, 2021-09-12:
So why do you use a spiffs partition? Try to recreate the file system like _boot.py does with:

import inisetup
vfs = inisetup.setup()

Friday8229, 2021-09-12:
I made the spiffs partition to store my CSV file and then read it from there with my C++ code. I tried the commands, below is the output.

>>>import inisetup
>>>vfs = inisetup.setup()
Traceback (most recent call last):
    File "<stdin>", line1 in <module>
    File "inisetup.py", line 34 in setup,
    File "inisetup.py", line 6 in check_bootsec,
AttributeError "NoneType" object has no attribute 'ioctl'
>>>

robert-hh, 2021-09-12:
Then try the code from inisetup.py, which will create a lfs2 file system. That is however not readable by code, expecting a spiffs file system.

import uos
from flashbdev import bdev

uos.VfsLfs2.mkfs(bdev)
vfs = uos.VfsLfs2(bdev)
uos.mount(vfs, "/")

Friday8229, 2021-09-12:
Hey thanks for responding, all I want to do is upload a file in the storage so that o can access it from the c++ code, there should be a way to do that ? What if I modify the partition table ? Or somehow get the vfs partition? I got no idea so any help is appreciated.

robert-hh, 2021-09-12:
The vfs partition must have the name vfs (not storage). Once you have that, MP should create the file system during boot. That one if accessible by MicroPython, but is not understood by your C code. So either you change your code in reading from a lfs file system, or you have to define a different interface for transferring the data.
@robert-hh

Friday8229, 2021-09-12:
Thanks, I'll change the partition name to vfs and try writing a code that works with littlefs. I saw [this](https://github.com/xreef/SimpleFTPServer) while searching, what if I convert the sketch provided at the end to a bin file and upload the file just like I flash the other bin files ?

robert-hh, 2021-09-12:
Using esptool.py, you can also write raw data to the flash at a fixed address, and use your C++ code to retrieve it from that place.
Since this is however not related to MicroPython, you should close the issue here.

Friday8229, 2021-09-13:
Thanks for your help, I have one last question. From what I read about mpy the instructions state that we should create the mpy file with the make command and then place it in a place from where it can be imported! So, where I should keep my mpy file before running the IDF.py / make command to make the micropython env? The board is esp32.

robert-hh, 2021-09-13:
That instruction seems odd and outdated. ESP32 uses a manifest.py structure, which points to the places where the .py files are located, which should be included. The default starting point is boards/manifest.py. One of the places is the esp32/modules directory. Any .py files located (or linked into) there will be compiled during the build process to a .mpy file and included as frozen bytecode.

peterhinch, 2021-09-13:
[This para](https://github.com/peterhinch/micropython-samples/tree/master/fastbuild#frozen-modules-and-manifests) is a brief introduction to frozen directories and manifests.

Friday8229, 2021-09-16:
Thanks a lot for all your support, I was able to use the make command successfully. Now my code runs well in the unix port but when I flash it on board there's one error. Btw in skipped the file reading part and stored the data in the vector directly.

Here's the error: 

```
$ picocom -b 115200 /dev/ttyUSB0
picocom v3.1

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready

>>> import cppexample
>>> cppexample.cppfunc()

abort() was called at PC 0x402acc26 on core 0

Backtrace:0x400912c6:0x3ffce1d0 0x40091989:0x3ffce1f0 0x400956ae:0x3ffce210 0x402acc26:0x3ffce280 0x40396985:0x3ffce2a0 0x402a93ce:0x3ffce2c0 0x402a914e:0x3ffce2f0 0x402a8d92:0x3ffce320 0x402a87f9:0x3ffce350 0x402a82d2:0x3ffce390 0x402a7c35:0x3ffce3d0 0x402af51d:0x3ffce680 0x402b55f9:0x3ffce6a0 0x402b5726:0x3ffce6c0 0x402b77e1:0x3ffce6e0 0x402af6f0:0x3ffce780 0x402b55f9:0x3ffce7d0 0x402b5622:0x3ffce7f0 0x402e1847:0x3ffce810 0x402e1b99:0x3ffce8a0 0x402c69b0:0x3ffce8e0

ELF file SHA256: d4430e51e3c3bc8b

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:5572
load:0x40078000,len:12696
load:0x40080400,len:4292
entry 0x40080414
MicroPython v1.16-243-g8c4ba575f-dirty on 2021-09-16; ESP32 module with ESP32
Type "help()" for more information.
>>> 

```

Friday8229, 2021-09-16:
ok so below is my code that I'm uploading. The error pops up when i store too many values in the vector around 18k to be specific.

 ```
#include <stdio.h>
#include "stdlib.h"
#include <stdlib.h>
#include <vector>

//fft start
using std::vector;

extern "C"{

#include <examplemodule.h>

}

mp_obj_t cppfunc() {
  vector<vector<double> > csv_data{ { "the numbers go here" } };
  
  mp_printf(&mp_plat_print, "CSV:%d\n", csv_data);
  return mp_const_none;
  
  }

 
```

When I run it from unix folder: 

```
$:~/micropython/ports/unix$ ./micropython
MicroPython v1.16-246-ga3ce8f08e-dirty on 2021-09-16; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import cppexample
>>> cppexample.cppfunc()
CSV:849142456
>>> 

```

When I upload it in the board:

```
:~/esp-idf/micropython/ports/esp32$ picocom -b 115200 /dev/ttyUSB0
picocom v3.1

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready

>>> import cppexample
>>> cppexample.cppfunc()

abort() was called at PC 0x400d3d33 on core 0

Backtrace:0x400912c6:0x3ffccd40 0x40091989:0x3ffccd60 0x400956ae:0x3ffccd80 0x400d3d33:0x3ffccdf0 0x401bbf7d:0x3ffcce10 0x400d0ade:0x3ffcce30 0x400d0926:0x3ffcce60 0x400d073e:0x3ffcce90 0x400d045d:0x3ffccec0 0x400d01fe:0x3ffccf00 0x400d0096:0x3ffccf40 0x400d6359:0x3ffccfa0 0x400dc435:0x3ffccfc0 0x400dc562:0x3ffccfe0 0x400de619:0x3ffcd000 0x400d652c:0x3ffcd0a0 0x400dc435:0x3ffcd0f0 0x400dc45e:0x3ffcd110 0x40108563:0x3ffcd130 0x401088b5:0x3ffcd1c0 0x400ed7dc:0x3ffcd200

ELF file SHA256: 37361dec047d77fd

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:5572
load:0x40078000,len:12696
load:0x40080400,len:4292
entry 0x40080414
MicroPython v1.16-243-g8c4ba575f-dirty on 2021-09-16; ESP32 module with ESP32
Type "help()" for more information.
>>> 
Terminating...
Thanks for using picocom

```

dpgeorge, 2021-09-19:
The original issue here about mpremote failing, was due to there being no filesystem on the esp32.

For the user C++ module, that's a separate issue and actually best to discuss in the forum because it doesn't look like a MicroPython issue.  It's probably a stack overflow.

closed 2021-09-19 by dpgeorge