micropython/micropython #7786: Concatenation of f-strings throws syntax error

peterhinch, 2021-09-14:
From [this forum thread](https://forum.micropython.org/viewtopic.php?f=2&t=11114) - ignore incorrect title:
```python
x = 1
y = 2
f"{x}" f"{y}"
```
On CPython this produces `12`. On MicroPython Unix build:
```python
MicroPython v1.17 on 2021-09-02; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> x = 1
>>> y = 2
>>> f"{x}" f"{y}"
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> 
```

dpgeorge, 2021-09-14:
This is a known limitation of f-strings in MicroPython.  It's actually documented: http://docs.micropython.org/en/latest/genrst/core_language.html#f-strings-don-t-support-concatenation-with-adjacent-literals-if-the-adjacent-literals-contain-braces

@jimmo it seems there's a minor issue with that cpydiff, that `"aa" f"abc"` is not supported in CPy either.

peterhinch, 2021-09-14:
Apologies. I hadn't spotted that doc update.

closed 2021-09-14 by peterhinch