micropython/micropython #7787: Why has MP lost bytes.decode() and str.encode(), breaking code?

peterhinch, 2021-09-14:
The loss of these methods has even broken [micopython-lib](https://github.com/micropython/micropython-lib/issues/443). While I agree there is a better (supported) way to do these conversions, breaking code seems unfortunate.

dpgeorge, 2021-09-14:
As far as I know there was no config change which disabled these.  Can you remember when it did work?

peterhinch, 2021-09-14:
No, but it certainly used to. Were these abandoned when `str(b'rats', 'utf8')` and `bytes('rats', 'utf8')` were introduced? Just a guess, mind.

jimmo, 2021-09-15:
@peterhinch Which port/version are you using that doesn't have these methods?

It's working for me on PYBv11. The only port that doesn't enable them is samd and zephyr (I'm not entirely sure why -- see #7655 for work to solve this).

peterhinch, 2021-09-15:
This is odd. Two weeks ago I needed to process a large collection of files which had been created with MicroPython pickle over 5 years. I used the Unix build. This threw an exception as detailed in https://github.com/micropython/micropython-lib/issues/443 which includes the exact build. I have since updated my Unix build to 
```
$ upython 
MicroPython v1.17 on 2021-09-02; linux version
```
and the fault has gone away. V1.17 on a Pyboard D SF2W is also correct.

So I guess we can close both issues, but I am puzzled as to how this happened.

closed 2021-09-15 by peterhinch