micropython/micropython #7794: umqtt not being built locally for esp32

pequnio3, 2021-09-14:
I am building micropython master locally and then flashing it to my esp32, however no umqtt module exists.

```
ImportError: no module named 'umqtt'
```

Build Command
```
cd ports/esp32
. ~/Documents/code/esp-idf/export.sh
```

https://github.com/micropython/micropython/issues/7185 seems to address the issue for the pre-built binaries but I am looking to make sure it is included in locally built binaries.

dpgeorge, 2021-09-15:
You'll need to build using the `manifest_release.py` file, like this:
```
$ cd ports/esp32
$ make FROZEN_MANIFEST=$(pwd)/boards/manifest_release.py
```

pequnio3, 2021-09-15:
Awesome Thanks!

pequnio3, 2021-09-15:
Tried that, but now receiving an error:

```
Traceback (most recent call last):
  File "/Users/blake/Documents/code/micropython/tools/makemanifest.py", line 177, in get_timestamp
    stat = os.stat(path)
FileNotFoundError: [Errno 2] No such file or directory: '/Users/blake/Documents/code/micropython-lib/python-ecosys/urequests/urequests.py'
```

Is there a preparation command I should have run?

JohnieBraaf, 2021-09-15:
Try donwloading this repo as well: https://github.com/micropython/micropython-lib

closed 2026-05-01 by dpgeorge