micropython/micropython #7795: PPP troubling with large messages > 1.4kB (esp32)

chtinnes, 2021-09-14:
I have a PPP connection with a SIM7000E module and just realized that large messages don't reach the target.
In my case, I want to send "large" messages via MQTT. From a size of around 1.4kB, the messages don't arrive at the MQTT broker anymore. I tested the same messages with WiFi and could send messages of up to 4.4kB. Unfortunately, I do not get any Error  which tells me what might go wrong.

To me, this looks like a limit due to some buffer. Is it possible to increase this buffer or is there another alternative to handle large messages?

Tangerino, 2022-06-01:
Hi, do you have a PPP sample code to share?
I can't make it to work. Tried the forum but no replies.

Tangerino, 2023-06-20:
Hi, I never manage to make PPP work, I'm using MP V1.19 without any customization. What do you use? Is there a code you can share that make PPP comes UP? Thank you in advance.