micropython/micropython #7798: pwm caused crash on esp8266 generic version

yunfan, 2021-09-15:
mpy ver: v1.17 generic esp8266
board: dmini

description: 

the board has a builtin rgb light, which could be control by pin (15, 12, 13)
i want it to do blink over and over, but the code only work for about 2 or 3 times, then the whole program will crash and reboot

here is the code in boot.py

```
import time
from machine import Pin, Timer, PWM

print("boot start")

def rgbBlink(rv, gv, bv, onTime):
    r, g, b = PWM(Pin(15)), PWM(Pin(12)), PWM(Pin(13))

    r.init(duty=rv)
    g.init(duty=gv)
    b.init(duty=bv)

    time.sleep_ms(onTime)

    r.deinit()
    g.deinit()
    b.deinit()
    
print("before doBlink")
def doBlink():
    tim = Timer(-1)
    tim.init(
            period=3000,
            mode=Timer.PERIODIC,
            callback=lambda t: rgbBlink(255, 128, 128, 1500)
    )

doBlink()
print("after doBlink")

gc.collect()
```

dpgeorge, 2021-09-15:
Thanks for the report, I can reproduce the issue (it crashes for some reason in the timer callback).

As a workaround I'd suggest constructing the PWM objects only once, outside the callback handler.

yunfan, 2021-09-16:
> Thanks for the report, I can reproduce the issue (it crashes for some reason in the timer callback).
> 
> As a workaround I'd suggest constructing the PWM objects only once, outside the callback handler.

thanks for the suggest, if i constructing the PWM objects outside, should i still deinit them in the callback handler?

dpgeorge, 2021-09-16:
>  if i constructing the PWM objects outside, should i still deinit them in the callback handler?

Only if you want to turn of the PWM output.

dpgeorge, 2021-09-16:
Or you could do `pwm.duty(0)` to turn them off (instead of deinit).

yunfan, 2021-09-16:
> Or you could do `pwm.duty(0)` to turn them off (instead of deinit).

thanks, this solved my problem finally, and i had checked your website, the cnc is cool

dpgeorge, 2021-09-16:
I'm glad you found a workaround.  But this is a real bug that needs to be fixed, so I'll reopen the issue.