micropython/micropython #7808: Minor Viper Flaw: Misleading error message

RaulKompass, 2021-09-16:
Accessing an undefined variable in Python and Micropython:
```
def quot(a:int, b:int) -> int:
    return c//b
```
results in `NameError: name 'c' isn't defined` during **execution**.

Above function with `@micropython.viper` results in `ViperTypeError: return expected 'int' but got 'object'`during **compilation**.
Just now I realize that viper has to assume that c is some global python object. Which means the error message is correct. But it is misleading.
An addition like  `name 'c' isn't defined while compiling` would be helpful imho.

robert-hh, 2021-09-16:
That's not a flaw, it is regular Python behavior. Object referenced in a statement do not have to exist before that statement is executed. So it does not have to be defined beforehand. If an object is not defined in a function, it is assumed to be a global object. Viper code can generally deal with Python objects, but does not accept mixed expressions.

jimmo, 2021-09-20:
> An addition like `name 'c' isn't defined while compiling` would be helpful imho.

@RaulKompass I don't think we could add this while still supporting arbritrary expressions (such as global variables) to be used in viper functions. The compiler doesn't have any concept of what is "defined" other than local variables.

rkompass, 2021-09-20:
@jimmo `ViperTypeError: return expected 'int' but got 'object' assuming c is global ` would be helpful for beginners like me. Now that I've realized the compilation assumptions, I will know I forgot something when I will see the `expected 'int' but got 'object'` in the future. That's the learning curve I suppose.
By the way: During programming graphical framebuffer functions I observed several advantages but also obstacles that limit the theoretical performance gains with viper that I would like to share/discuss. I have some knowledge of C but I don't even have an idea where in the micropython kernel code the compilation of viper is happening. So I can formulate my ideas for improvement only as questions or enhancement proposals. But I don't want to complain. Working with micropython is a pleasure. Where would be the best place to report my findings? Here as a new issue (which it not really is)? Or in the forum?