micropython/micropython #7816: V1.17 threading versions Error running reset production mode

Spring641, 2021-09-18:
In the thread version of the latest firmware V1.17, pressing the USR button to reset the system after the system starts will cause the system to crash! According to my current debugging code, I found that the execution of mp_hal_delay_ms() in the thread version code will cause the system to crash.

CaptainJackey, 2021-09-18:
v1.16 is ok, v1.17's thread has bug. 

version : threading: v1.17-47-gb326edf68 (latest)

**Factory reset the filesystem invalid ， please cheak!**

IlysvlVEizbr, 2022-01-18:
In v1.18 the bug is still exist.
Safe mode & Factory reset the filesystem function of pyboard cannot be used

CaptainJackey, 2022-01-18:
> o

Yes, For the threading firmware.  Normal firmware is OK! @dpgeorge

jimmo, 2022-08-25:
I was able to repro this -- the threading firmware seems fine and passes the unit tests, but the safe-mode boot sequence is failing.

You're right, the culprit is `mp_hal_delay_ms`. PR incoming.

commit 986ad6bf (2022-08-26): stm32/boardctrl: Use HAL_Delay instead of mp_hal_delay_ms.

Not safe to use mp_hal_delay_ms before boot if threading is enabled,
because threading will not have been initialised, and
MICROPY_EVENT_POLL_HOOK assumes threading is initialised.

HAL_Delay doesn't call MICROPY_EVENT_POLL_HOOK, but is still
power-efficient like mp_hal_delay_ms (unlike mp_hal_delay_us).

Fixes #7816.
closed 2022-08-26 by dpgeorge