micropython/micropython #7819: Problems of esp32, io16 and io17 in spiram mode

wangshujun-tj, 2021-09-19:
In spiram mode, io16 and io17 are used for CS and CLK of spiram. If these two pins are set in MPY program, MPY will crash.

It is recommended to disable these two pins in the pin initialization of esp32.

My tests are as follows:

    {{&machine_pin_type}, GPIO_NUM_14},
    {{&machine_pin_type}, GPIO_NUM_15},
    #if CONFIG_ESP32_SPIRAM_SUPPORT
    {{NULL}, -1},
    {{NULL}, -1},
    #else
    {{&machine_pin_type}, GPIO_NUM_16},
    {{&machine_pin_type}, GPIO_NUM_17},
    #endif
    {{&machine_pin_type}, GPIO_NUM_18},

UnexpectedMaker, 2021-09-20:
This is a good idea, thanks @wangshujun-tj . I've made this change as part of some other work I am doing in `machine_pin.c` and will do a PR later today.

UnexpectedMaker, 2021-09-20:
Ok, I've done a PR, so feel free to close this issue @wangshujun-tj :)
https://github.com/micropython/micropython/pull/7821

wangshujun-tj, 2021-09-20:
Commendable speed

closed 2021-09-20 by wangshujun-tj

petrkr, 2021-12-01:
This breaks SPIRAM on other pins than most of board uses. Check #8028