micropython/micropython #7820: ESP32-S2/S3: CDC stalls sending serial output on boot/hard reset.

UnexpectedMaker, 2021-09-20:
On the ESP32-S2/S3, the CDC long stalls on boot or on a hard reset, listening for a serial connection.
If no connection is present, the more serial output the code has ( `print('lots of stuff')` ), the longer the stall as its waiting for 1 second per flush, per chunk of serial output.
https://github.com/micropython/micropython/blob/782d5b2e534c96f4668443efe37c0514e8fcfe91/ports/esp32/usb.c#L86

We need a way to have the USB determine there is no serial connection, and ignore sending and flushing of the output.

@dpgeorge this is what you, Matt and Jimmo and I were talking about last weekend. I'm just creating an issue for it so it doesn't get forgotten :)

wangshujun-tj, 2021-09-20:
This is my attempt. The method is not perfect. After USB connection, DTR needs to be set to respond to the communication of USB port

However, there are no other clear instructions for connection and disconnection in IDF. According to the Convention, DTR is a signal indicating that the equipment at the opposite end is ready, which is reasonable, but many USB tools do not follow this Convention.
[usb.zip](https://github.com/micropython/micropython/files/7193448/usb.zip)

wangshujun-tj, 2021-09-20:
The modification I made is that the default serial port is always open. You can test the way that two serial ports work at the same time

The input response from any one port is sent to both ports at the same time. After disconnecting the USB, the first character of the serial port will delay for about 1 second, and then continue to work

UnexpectedMaker, 2021-09-20:
Have you tried this on hardware?

UnexpectedMaker, 2021-09-20:
It seems that if DTR and RTS are both true, there is a connected device, otherwise there isn't, according to this:
https://github.com/espressif/esp-idf/blob/220590d599e134d7a5e7f1e683cc4550349ffbf8/components/tinyusb/additions/src/tusb_cdc_acm.c#L57

So maybe we just need...
```
void usb_callback_line_state_changed(int itf, cdcacm_event_t *event)
{
    int dtr = event->line_state_changed_data.dtr;
    int rst = event->line_state_changed_data.rts;
    ESP_LOGI(TAG, "Line state changed! dtr:%d, rst:%d", dtr, rst);
    if ( dtr && rts) {
        usb_cdc_linked=1;
    }else{
        usb_cdc_linked=0;
    }
}
```
and

```
void usb_tx_strn(const char *str, size_t len) {
    while (len) {
        size_t l = len;
        if (l > CONFIG_USB_CDC_TX_BUFSIZE) {
            l = CONFIG_USB_CDC_TX_BUFSIZE;
        }
        if (usb_cdc_linked==1) {
            tinyusb_cdcacm_write_queue(CDC_ITF, (uint8_t *)str, l);
            (tinyusb_cdcacm_write_flush(CDC_ITF, pdMS_TO_TICKS(1000)) != ESP_OK)
        }
        str += l;
        len -= l;
    }
}
```

UnexpectedMaker, 2021-09-20:
Ok, the above works on S2 hardware :)
I'll put together a PR for it now... thanks again @wangshujun-tj !

wangshujun-tj, 2021-09-20:
I first judged RTs and DTR at the same time, but according to the definition of RS232, I think it is more reasonable to remove RTs
These codes have been tested on the hardware. At the same time, the modification items always initialize the serial port, and the sending information is always sent to the serial port

wangshujun-tj, 2021-09-20:
The test circuit board I used is esp12k, with an 8-megabit spiram esp32s2 module

UnexpectedMaker, 2021-09-20:
The IDF examples use both DTR and RTS, so I am using both in my implementation, and it's working as expected on 2 of my S2 boards. If there is an issue with using g both RTS & DTR, then I am able to put it back on Espressif to sort out :)

My PR is here...  https://github.com/micropython/micropython/pull/7825
Feel free to give it a go on your hardware.

I'm going to close this issue now and wait to see if @dpgeorge is happy with the implementation.

closed 2021-09-20 by UnexpectedMaker

wangshujun-tj, 2021-09-20:
There is no problem judging two at the same time.
According to the signal definition, if only DTR is judged, the connection will be triggered as long as the terminal is enabled. At the same time, it is judged that the connection will not be triggered until the terminal sends it for the first time.
There is no obvious difference in actual use. I use an overly free serial port tool, so it is inconvenient to make judgment.
DTR（Data Terminal Ready）
RTS（Request To Send）