micropython/micropython #7824: An attempt on S2 / C3 hardware SPI

wangshujun-tj, 2021-09-20:
There are two modifications:
For the setting of S2 / C3 DMA channel, the parameter changes to 3 and is automatically allocated
S2 add definition of fspi
In the preliminary test, th
[machine_hw_spi.zip](https://github.com/micropython/micropython/files/7193440/machine_hw_spi.zip)
e SPI transmission is normal on S2 / C3, and the reception is not tested

UnexpectedMaker, 2021-09-20:
Sorry, I'm not suer what this is fixing.... I added HW SPI for S2 a few days back and it was merged on the weekend.
Looks like your code is on top of some older code.

Check my PR:
https://github.com/micropython/micropython/pull/7814

wangshujun-tj, 2021-09-20:
I made the changes on the 1.17 release. There were these problems in the distribution at that time.
After reading the code you released, I initially suspect that 1. The SPI problem of C3 is not solved, 2. There may be a problem with the fspi DMA channel
I'll actually compile the test as soon as possible

UnexpectedMaker, 2021-09-20:
I'm not testing C3 stuff - I didn't add that part, but the S2/S3 stuff works correctly, and the DMA channels are correct on the S2, unless you can give me some code that shows it not working correctly. 

> I made the changes on the 1.17 release. There were these problems in the distribution at that time.

You should always be hacking away against latest master, not a release tag :)

wangshujun-tj, 2021-09-20:
After testing, there is no problem with the DMA channel of fspi. You set 1 and I set 3. It seems that both are correct. The IDF document does not seem to have a clear input requirement
I suggest you take the time to check my code. The amount of changes is very small and can be compatible with C3

UnexpectedMaker, 2021-09-20:
> After testing, there is no problem with the DMA channel of fspi. You set 1 and I set 3. It seems that both are correct. The IDF document does not seem to have a clear input requirement
> I suggest you take the time to check my code. The amount of changes is very small and can be compatible with C3

Well please refactor your changes agains the latest in master, so I can identify it easier. Thanks :)

closed 2021-09-28 by wangshujun-tj