micropython/micropython #7842: Support WizNet 5100 for STM32 board

vinz-uts, 2021-09-22:
Hi, this is not a really issue. I needed to get work an external shield with WizNet 5100 driver with stm32 boards. I add the `w5100` folder in `drivers/wiznet5k/ethernet` (modifing the original driver files for compliance with micropython) directory and I've done some changes in `ports/stm32/modnwwiznet5k.c'. Everything seems to work. I set the network configuration and read from a ntp server the current time. I don't know how share my code if someone needs it, if I've to open a pull request or do a simple commit in master branch. All need more accurate tests.
This is all.
Thanks for micropython project.

vinz-uts, 2021-11-15:
Hi, I've also modified a pure-python driver for the Wiznet 5500 (originally writed from Adafruit for CircuitPython) that seems to work with ESP32. If anyone tell me how, I can add these to the repository in the right folder.
@dpgeorge 
Best Regards.

bartagergely, 2021-12-28:
Hi @vinz-uts 

Unfortunately I am not the one who can help you with the "How can I contribute question", but I'm interested in both of your solutions (wiz5100 on stm32, and wiz5500 on esp32). Can I get the modifications (and some brief instructions) somehow?

vinz-uts, 2021-12-29:
Hi, I'm glad to be able to help you. To include the Wiznet 5100 drivers, I've edited some files in `ports/stm32/` folder: `modnwwiznet5k.c` and `network_wiznet5k.c`, next I've added the `W5100` folder in `drivers/wiznet5k/ethernet`. I've also modified the files `socket.c`, `wizchip_conf.c` and `wizchip_conf.h` in the same folder. I'm using this wiznet5k driver on my custom board based on stm32 chip without probem, I hope it also works in your application without problems.
To use the ethernet driver I use the following lines:
```python
spi = SPI(2)
nic = network.WIZNET5K(spi, Pin.board.PB12, Pin.board.PC4)
nic.ifconfig(('192.168.1.18', '255.255.255.0', '192.168.1.1', '8.8.8.8'))
```
[wiznet5100.zip](https://github.com/micropython/micropython/files/7789199/wiznet5100.zip)

For esp32 I wasn't able to modify the low level driver, so I've edited some pure-python driver borned for adafruit cyrcuitpython. In the `wiznet5k` there are all the necessary files, including a test file (`esp32_eth_wiznet5500_ntp.py`) that queries a ntp server.

[wiznet5k.zip](https://github.com/micropython/micropython/files/7789255/wiznet5k.zip)

I hope these can help you.
Best regards.

bartagergely, 2021-12-29:
@vinz-uts 
Thank you for the files and the description. I am now working on a energy monitoring and acting solution based on modbus inter-communication, and a webpage for human interaction, for one of my hobby projects. (An amateur weather and environment monitoring station, powered by solar panel, wind generator and batteries.)
I have STM32s, ESP32s, PiPicos, W5500s, W5100s, and now I'm experimenting which combination of these will be the optimal choice for this task. 
Your solutions will help me a lot in this process, thank you again.

vinz-uts, 2021-12-29:
Nice project, I hope these files will help you. Give us feedback if they work.

projectgus, 2024-07-05:
I'm going to close this issue out. There is a standard Wiznet 5K driver that works across multiple ports including STM32:
https://docs.micropython.org/en/latest/library/network.WIZNET5K.html

The driver does not currently compile with `MICROPY_PY_NETWORK_WIZNET5K=5100` as it uses a function that only work on the later chips not 5100, but fixing this would be the way forward if someone wanted to add official support in the future.

closed 2024-07-05 by projectgus