micropython/micropython #7847: PEP 634 Structural Pattern Matching (Python 3.10) support

nickovs, 2021-09-23:
CPython 3.10 release candidate 2 is now out, and the final release is less than two weeks away. By far the biggest [change in the release](https://docs.python.org/3.10/whatsnew/3.10.html) is support for the new Structural Pattern Matching syntax. This is by far the most substantial language change Python has seen since the arrival of the walrus operator `:=`, and probably since the release of Python 3.

(Those not familiar with Structural Pattern Matching should probably start by reading [PEP 635](https://www.python.org/dev/peps/pep-0635/) which covers the motivation for this new language feature. There are separate PEPs for the [full specification](https://www.python.org/dev/peps/pep-0634/) and a [tutorial](https://www.python.org/dev/peps/pep-0636/).)

Structural Pattern Matching is a big and fairly complex feature; adding it to Micropython would be a fair amount of work and add quite a bit of code to the parser. How much work and how much code are open questions, as is the question of when, if ever, it will be worth doing? The answer to this last question will depend very much on how rapidly users of CPython embrace, and come to depend on, this new feature. That said, the new syntax can save a lot of source code typing; if it gets rapid and windspread uptake it would represent a substantial incompatibility between Micropython and CPython.

Given how significant an addition this is to the Python syntax, it seems like it would be a good idea to understand the work involved and the scope of the change well before use of it in CPython becomes widespread. I am opening this issue to serve as a place to discuss those topics.

mattytrentini, 2021-10-15:
Structural Pattern Matching is certainly a powerful feature but it's a _challenging_ addition to MicroPython since it was one of the first CPython features that _required_ the PEG parser. PEP 635 mentions this in [Backwards Compatibility](https://www.python.org/dev/peps/pep-0635/#backwards-compatibility): "...3rd party tooling that uses a LL(1) parser to parse Python source code may be forced to switch parser technology to be able to support those same features"

PEG parsers provide a number of improvements over LL parsers but one of the consequences is that their maximum memory use is typically _significantly_ higher (because their look-ahead buffer can be almost unconstrained). That's a reasonable cost on the desktop but in the embedded domain it will be difficult to manage.

Not saying it's impossible but, yes, very _challenging_!

moefear85, 2023-06-12:
i just learned of this feature, am overjoyed with the simplification potential it provides, and from [this](https://stackoverflow.com/questions/75188852/match-case-in-micropython-syntaxerror-invalid-syntax) discussion I assume I'm not alone

mattytrentini, 2023-06-13:
> Btw... just let ENOMEM decide what one can or cannot match, not prohibit it to begin with. If anything, the feature
> can be made to require psram and fall back to LL when there is none, then raise unsupported PEP exception on
> a match statement

It's not _just_ a memory issue. Structural pattern matching is implemented with so-called _soft keywords_. Specifically, 'match' and 'case' are now keywords but can also be used as identifiers (necessary for [backwards compatibility](https://peps.python.org/pep-0635/#backwards-compatibility)). Not easy with an LL parser, particularly with a constrained look-ahead buffer. 

Consider this now valid python code (from [An overview of structural pattern matching for Python](https://lwn.net/Articles/893193/)):

```python
    match = re.match(PATTERN, "...")
    match match:
	case case, "closed":
	    ...
```

I concede it _may_ be possible to extend the LL parser to support such syntax - and I encourage folks to raise PRs to do so and prove me wrong! 😜

moefear85, 2023-06-13:
don't be hard on yourself. it's ok if you don't know how to do it. many tutorials on the internet explain how meaning depends on the position.

btw, backwards compatibility isn't any more necessary than forwards, so leaving the feature out so as not to compromise that is a weak point. better would be an option that configures which parser to use, keeping LL as default. then no backwards compatibility issue.