micropython/micropython #7856: RP2: Crash when using threading with import

RcSepp, 2021-09-25:
There seems to be a bug that causes the RP2 board to crash when using threading in combination with imports (of user-provided modules).
(Note: I also [posted this issue in the forum](https://forum.micropython.org/viewtopic.php?f=2&t=11122), but it seems this is the better place.)

### Test setup
1. Connect a Raspberry Pi Pico with firmware image `rp2-pico-20210902-v1.17.uf2` via USB.
2. Launch Thonny in MicroPython (Raspberry Pi Pico) mode.
3. Create an empty file and store it in Pico's root directory as "some_module.py"
4. Run the following code. -> The Pico will crash (usually in line `for bar in foo.values()` after ~2 iterations).
5. Reset the Pico by unplugging and replugging the USB cable and pressing the STOP button (CTRL-F2).
6. Remove the line `import some_module`.
7. Run the code again. It will now finish successfully.

```
import time, _thread
import some_module

def some_thread():
	for i in range(10):
		print(i)
		time.sleep(0.5)

_thread.start_new_thread(some_thread, ())

foo = {}
while True:
	for bar in foo.values():
		pass
```

### Scoping down the issue
* I'm not sure whether this is a board-specific or general problem, because I only have access to a Raspberry PI Pico.
* I tested the above on 2 different Pico's, so I don't think it's a hardware glitch.
* I tested a slightly modified version (with a blinking led) without Thonny (using rshell) and the Pico still crashed, so I don't think the problem comes from Thonny either.

kdschlosser, 2021-10-14:
I don't have this issue on an ESP32 at all. so it is probably port specific.

I have a silly question tho.
have  you tried changing your imports around.

```python
import time
import _thread
import some_module
```

```python
import some_module
import time
import _thread
```

I personally have never liked using a "," on an import statement unless it is a `from a import (x, y, z)` type of statement. And I always wrap the imported values in "( )" if I am importing more then a single item.

see if doing the imports all on separate lines fixes it or moving the user module import to the top or bottom fixes it.

kdschlosser, 2021-10-14:
also if you can provide the exact output of the crash that would be helpful as well.

kdschlosser, 2021-10-16:
The import is not taking place in a different thread.

If you were doing this then I would say there is something wrong with the import mechanics. 
```python

import time, _thread

def some_thread():
    import some_module
    for i in range(10):
        print(i)
        time.sleep(0.5)

_thread.start_new_thread(some_thread, ())

foo = {}
while True:
    for bar in foo.values():
        pass
```

However. I am a ding bat and didn't see  this before.

```python
while True:
    for bar in foo.values():
        pass
```

There is your deadlock. You are spinning wheels. this is going to chew up all of the processor time and it could be causing the crash.

try this code

```python
import time, _thread
import some_module

def some_thread():
    for i in range(10):
        print(i)
        time.sleep(0.5)

_thread.start_new_thread(some_thread, ())
```

MicroPython does not work like CPython does. The interpreter never exits with MicroPython it is locked into running.So you don't have to hold up the main thread at all.

Just to let you know that if you did want to stall a thread you would need to use a lock to do that.
```python
import time, _thread
import some_module

def some_thread():
    for i in range(10):
        print(i)
        time.sleep(0.5)

_thread.start_new_thread(some_thread, ())
lock = _thread.allocate_lock()
lock.acquire()

# the second acquire causes the thread to get blocked, this will not cause a 
# "spinning wheels" situation where processor time gets all used up. 
lock.acquire()

```