micropython/micropython #7858: Implicit f-string concatenation is flawed.

gmos, 2021-09-25:
```
# Following implicit f-string concatenation should work.  It does in cPython.
>>> x = (
...   f"{dir()}"
...   f"{dir()}"
... )
Traceback (most recent call last):
  File "<stdin>", line 4
SyntaxError: invalid syntax

# Following does work syntactically, but should treat the second string as unformatted.  It seems to do some sort of dictionary look-up.
>>> x = (
...   f"{dir()}"
...   "{dir()}"
... )
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
KeyError: dir()
>>>
```

gmos, 2021-09-25:
Workaround is simple in both cases: make the concatenation explicit by adding a "+"

dlech, 2021-09-25:
Is this different from the already documented differences? 

https://docs.micropython.org/en/latest/genrst/core_language.html#f-strings-don-t-support-concatenation-with-adjacent-literals-if-the-adjacent-literals-contain-braces

gmos, 2021-09-25:
Nope.  Thanks @dlech.

closed 2021-09-25 by gmos