micropython/micropython #7860: framebuf.FrameBuffer: not secured against silly/wrong/malicious strides

volkerjaenisch, 2021-09-26:
Dear Micropython Developers!

framebuf.FrameBuffer is not secured against the misuse with unreasonable values of the stride parameter.

IMHO :
- stride < width  is nonsense (or only useful for esoteric matters) and should be set to width
- strides larger than len(buf)/width are an attack vector to disclose information

So this can happen:

```
buffer = bytearray([0, 0, 0, 0, 0xff, 0x00, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff])

for stride in range(100):
    fb = framebuf.FrameBuffer(buffer, 2, 2, framebuf.MONO_HLSB, stride)
    a = fb.pixel(1,1)
    print(stride, a)
```

The function getpixel should only operate **in** the buffers first four bytes, thus only returning zero values. 
But the bit pattern of the wider buffer is exposed.

```
>>> %Run -c $EDITOR_CONTENT
0 0
1 0
2 0
3 0
4 0
5 0
6 0
7 0
8 0
9 0
10 0
11 0
12 0
13 0
14 0
15 0
16 0
17 0
18 0
19 0
20 0
21 0
22 0
23 0
24 0
25 1
26 1
27 1
28 1
29 1
30 1
31 1
32 1
33 0
34 0
35 0
36 0
37 0
38 0
39 0
40 0
41 1
42 1
43 1
44 1
45 1
46 1
47 1
48 1
49 1
50 1
51 1
52 1
53 1
54 1
55 1
56 1
57 1
58 1
59 1
60 1
61 1
62 1
63 1
64 1
65 1
66 1
67 1
68 1
69 1
70 1
71 1
72 1
73 1
74 1
75 1
76 1
77 1
78 1
79 1
80 1
81 1
82 1
83 1
84 1
85 1
86 1
87 1
88 1
89 0
90 0
91 0
92 0
93 0
94 0
95 0
96 0
97 0
98 0
99 0
```

This allows a malicious code to read arbitrary RAM positions.

To prevent this there should be checks for dangerous stride values while creating a FrameBuffer instance.

I noticed this while bringing up my [testing for the new FrameBuffer.get_rect() routine. ](https://github.com/Inqbus/micropython/blob/get_rect_raw/tests/extmod/framebuf1_get_rect.pyl)

Also there is not a single test in the existing framebuf testing code concerning the stride parameter.

Sorry for counting peas. 

Cheers,
Volker

peterhinch, 2021-09-27:
MicroPython provides numerous ways to access arbitrary memory locations. Built-in assembler for a start.

volkerjaenisch, 2021-09-28:
> MicroPython provides numerous ways to access arbitrary memory locations. Built-in assembler for a start.

Nearly any programming language has. So your statement is trivial and not a valid argument against security.

Security breaches do not occur because some idiot pipes Web-Requests (e.g. MQTT) unfiltered to the assembler. They occur when resoanable developers rely their  code on software that looks not likely to be security relevant, but is not secure: Like Framebuf. Who expects a framebuffer to be not securily confined? Not me.

A wrong stride setting can overwrite code. For you as a C-Developer this may not look like a terrible news. But the average Python developer switching from C-Python to MicroPython has never experienced a CoreDump while programing C-Python. Because C-Python is as secure as possible.

The unsecured stride parameter would cause a core dump on a system having a core. And since there is no RAM-Page security on MicroControllers the problem does not even pop up.

So what are you hinting here at? 
* That MicroPython is generally not secure - take it or leave it. 
* That security is an elaborate feature only needed on servers and the such and not for MicroControllers?

There are so many systems out there in the IoT and more and more are MicroControllers with WLAN/IPv6 and therefore an ideal target vector for hackers. Why hack the Webserver, Firewall of a big company if there are  IoT Microcontrollers measuring moisture and displaying this information via framebuf, which leads to a security breach that I can use to hack into for further probing or even attacks.

Printers are in a same category of vulnerability for more than two decades. 

IMHO it is our responsibility as developers to write code as secure as possible. May this be in OS, Application or MicroController code. There may have been a difference in times when MicroController were build into washing machines and not connected to more than the grid. But the IoT changed this so much.

Coming back to the initial issue. The security leak of framebuf can unobtrusively be dammed by a single additional term to accompany the already existing six validation terms.

`    if (h < 1 || w < 1 || x + w <= 0 || y + h <= 0 || y >= fb->height || x >= fb->width) {
`

So why is there a reason for discussion at all? It should simply be fixed.

Cheers,
Volker

stinos, 2021-09-29:
> Who expects a framebuffer to be not securily confined?

Running an application on an embedded device with a screen where a moving object goes to the side of the screen then just freezes because of an out of bounds access something I've seen more than just once, so I'm not going to expect much here, given my experience.

In Python I'd expect there to be bounds checking though, but depending on the level of the implementation; as hinted by Peter: if I'm working with something which even remotely looks like raw memory access I'm always going to inspect the documentation or implementation code to figure out whether it's guarded. So in this case I agree there should be bounds checking.

> That MicroPython is generally not secure - take it or leave it

Given the varied feature set and different components, it would be almost certainly incorrect to make general statements.

volkerjaenisch, 2021-10-01:
@stinos 
Thanks for your testimony.

Cheers,
Volker