micropython/micropython #7862: How use _thread library?

wangdy-code, 2021-09-27:
**### borad:esp32**

I expect to use this library to create a thread to detect whether a thread with the same function has been created and delete the previous thread with the same function.

kdschlosser, 2021-10-14:
you can't "delete" threads. I am I am not sure what you mean about function. I need a code example to see what you are wanting to do. It doesn't need to be functional code pseudo code is fine so I can get a better understanding.

mattytrentini, 2021-10-15:
You may want to watch the Melbourne MicroPython Meetup video from [November 2018](https://melbournemicropythonmeetup.github.io/November-2018-Meetup/) where Damien discusses the (then, recently added!) `_thread` module.