micropython/micropython #7865: Faulty exception: Implicit string conversion: bad_implicit_conversion(...)

Michael-dot-byte, 2021-09-27:
Dear MicroPython community,

I stumbled upon a false exception when trying to implicitly convert a string to integer.
Error message should be: "can't convert 'str' object to int implicitly" 
Rather than: "can't convert 'int' object to str implicitly" (see Traceback)

I'd be happy to see a fixture in future version or get any feedback of how to solve the issue.

Script:
=======
def func(input1, input2):
	return input1/input2

print(func(1,2))
print(func("1",2))

Result:
=======
Traceback (most recent call last):
  File "../implicit_conv_exception.py", line 7, in <module>
  File "../implicit_conv_exception.py", line 4, in func
TypeError: can't convert 'int' object to str implicitly

Details:
========
- Executed on Micropython v1.17
- Port: unix
- Methode causing error:  STATIC NORETURN void bad_implicit_conversion(mp_obj_t self_in)

dlech, 2021-09-27:
I would expect the error to be that string doesn't support the `/` operator.

stinos, 2021-09-27:
Well, CPython reports `unsupported operand type(s) for /: 'str' and 'int'` (or vice-versa when changing the order.

But since Python doesn't have implicit conversions between int and str anyway, it's in my opinion irrelevant which way the MicroPython error message reports it.