micropython/micropython #7867: Ctrl+A followed by Ctrl+C gets serial input stuck since 1.17

aivarannamaa, 2021-09-28:
OS: Ubuntu 20.04

Device: RPi Pico

main.py:

```python
from machine import Pin
from time import sleep
led = Pin(25, Pin.OUT)
n = 0

while True:
    led.toggle()
    print("13 x {} = {}".format(n, 13*n)) # print the thirteen-times table
    n = n+1
    sleep(0.5)
```

With MicroPython 1.17, when I plug this device in, connect with `tio` and press Ctrl+C, then it successfully interrupts and gives me normal prompt. When I, instead first press Ctrl+A and then Ctrl+C, then the input kind of gets stuck (if I do it via `pyserial`, then the write method blocks). Same happens when I press Ctrl+D and then Ctrl+C.

Sometimes I get out of this situation by hard-resetting Pico, but sometimes even this doesn't help and I need to either restart my computer or [reset USB](https://gist.github.com/planetceres/917840478e1e4d45f8373667630e51a0).

When I go back to 1.16, then this problem disappears.

This issue is relevant for Thonny IDE, because upon connecting a device, it sends Ctrl+A to probe whether the device is already idle at prompt or a process is running.

robert-hh, 2021-09-28:
Trying to replicate that with tio, Ctrl-A sets the device into Raw mode, which is terminated by Ctrl-B. Ctrl-C is ignored in raw mode. That seems consistent. Version 1.16 and 1.15 behave identical.

aivarannamaa, 2021-09-28:
> Ctrl-C is ignored in raw mode.

With all devices and MP versions I've seen, this is not the case.

aivarannamaa, 2021-09-30:
> Version 1.16 and 1.15 behave identical.

@robert-hh, do I understand correctly, that according to your experiments, Ctrl+A followed by Ctrl+C also stucks Pico (or something else?) in 1.16 and 1.15?

robert-hh, 2021-09-30:
@aivarannamaa In my test all versions behave identical in that in raw mode Ctrl-C is ignored and the device stays in raw mode, until it is terminated by Ctr-B. I conclude that the Pico is not stuck, because it still responses to Ctrl-B.

aivarannamaa, 2021-10-02:
> In my test all versions behave identical in that in raw mode Ctrl-C is ignored

This is really strange... I now tried once again Pico with MP 1.17 and Sparfun ESP32 Thing with MP 1.16:

```
>>> from time import sleep
>>> 
raw REPL; CTRL-B to exit
>OK1
1
1
1
1
1
1
1
1
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
KeyboardInterrupt: 
>
```

On the line witih `>OK1`, I blindly entered following code:

```
while True:
    print(1)
    sleep(0.5)
```

As you can see, I was able to interrupt without problems.

Am I really the only one who can interrupt programs in raw mode with Ctrl-C? Can someone else please try my experiment from the top of this thread? @aallan?

robert-hh, 2021-10-02:
Using your example, I can terminate the runing code for both versions with ctrl-C, after I terminate the code entry with Ctrl-D. While in Raw mode entry state, before entering Ctrl-D, Ctrl-C is ignored.

aivarannamaa, 2021-10-02:
Great! At least we're on the same page with this one.

Could you now please repeat this experiment with 1.17 on Pico but _after_ submitting the code with Ctrl-D and _before_ interrupting the code with Ctrl-C, you press Ctrl-A as if you were trying to enter raw mode (even though you are in raw mode already). Can you now interrupt the code with Ctrl-C?

robert-hh, 2021-10-02:
In that case I cannot interrupt the code with Ctrl-C, both for Version 1.17 and 1.16, and I cannot stop the mode with either Ctrl-D or Ctrl-B.
MicroPython v1.17-69-gbfb2f52a6
MicroPython v1.16-241-ga6bab54a5
After I terminated the Code with Ctrl-C, it is still in raw mode, and I can again enter code and run and interrupt it. But in the initial state after entering Ctrl-A, Ctrl-C is ignored.

aivarannamaa, 2021-10-02:
I see, in this case it would be good if someone else also makes this experiment, because I just confirmed once more that in 1.16, issuing Ctrl-A before Ctrl-C causes no problems, but does in 1.17.

In any way, in my opinion it's a bug, even if it's not clear whether it is a regression.

aivarannamaa, 2021-11-13:
Tried to reproduce the problem on Sparkfun ESP-32 with MicroPython 1.17 and Adafruit CP Playground with CircuitPython 7.1 beta (based on MP 1.17) but failed, ie. it looks like Pico-specific problem.

Also tried with the latest nightly build of MP 1.17 and Pico and could still reproduce it. 

Also, I found that you don't need to press Ctrl+A to mess up the possibility of interrupting with Ctrl+C later -- an ENTER would do as well.

The regression between MP 1.16 and 1.17 could be explained by the inclusion on Pico SDK 1.2. They recently released SDK 1.3. Is this being used in nightly builds already? @dpgeorge ?

scruss-elmwood, 2021-11-25:
Can confirm exact behaviour using Aivar's code on MicroPython v1.17 on 2021-09-02; Raspberry Pi Pico with RP2040: Pico appears to accept no more input after Ctrl+A and needs to be power cycled to get a usable REPL back

Having similar problems with a simple LCD clock that's locking out the Pico REPL after starting it in Thonny.

pdw-mb, 2022-01-21:
I can confirm the exact same issue here, and that the issue is still present in v1.18: pressing Ctrl-A whilst a script is running prevents Ctrl-C from interrupting that script.

The issue only seems to exist if you press Ctrl-A when code is running.  At the REPL prompt, pressing Ctrl-A then Ctrl-B is fine, as is Ctrl-A, Ctrl-C, Ctrl-B.  I can then run some code (e.g. `sleep(5)`) and Ctrl+C interrupts it:

```
>>> 
raw REPL; CTRL-B to exit
>
MicroPython v1.18 on 2022-01-17; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> time.sleep(5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt: 
>>> 
```

If instead, I run `sleep(5)` then press Ctrl-A, Ctrl-C, nothing happens: all further input is ignored until the sleep completes. 

On v1.16, Ctrl+A, Ctrl-C kills the running script:
```
>>> time.sleep(5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt: 
```

The issue was made worse for me by the fact Ubuntu ships with ModemManager which seemingly probes newly connected USB devices with Ctrl-A, with the effect that once I installed a `main.py` script that didn't exit, I could not regain control of the Pico other than by reinstalling v1.16, or using flash_nuke.uf2. I have now killed ModemManager.

bennybubble, 2022-03-02:
Facing the same problem, after uploading main.py there is NO WAY to Ctrl-C out of it.
Have to load a blank UF2 file and micropython (1.17) again to see get control over it.

Any solution on the horizon?

paul-ollis, 2022-04-08:
This has been fixed by https://github.com/micropython/micropython/commit/587339022689187a1acbccc1d0e2425a67385ff7.

aivarannamaa, 2022-04-09:
@paul-ollis, thank you for this information (I wasn't aware of #7996) and @hoihu, thank you for the fix!

It looks my problem doesn't exist in the latest nightly anymore. If someone else can also confirm this, then I guess this issue can be closed as well.

robert-hh, 2022-04-09:
The fix is for the rp2 port only. A PR fixing that for the mimxrt port is made. A samd port fix will follow.

scruss, 2022-04-11:
> [It looks my problem doesn't exist in the latest nightly anymore. If someone else can also confirm this, then I guess this issue can be closed as well.](https://github.com/micropython/micropython/issues/7867#issuecomment-1093776168)

Confirmed clear on `rp2-pico-20220410-unstable-v1.18-307-g66b5c4c77.uf2`: a running main.py does not lock up the REPL like before.

Thanks to all who contributed!

cruelke, 2022-05-01:
Hello
i have the same problem and tried also the latest and older versions without success.
Even with reseting the flash memory and using 2 Pico's .
Can anyone help how this can be solved ?

robert-hh, 2022-05-01:
That should be fixed in the daily builds since a while. They are called unstable, but in fact they usually are stable.