micropython/micropython #7871: ESP32C3 timer issue

bdambrosio, 2021-10-02:
When using the micropython ble repl example (works on regular esp32), I receive the following error:
    dupterm: Exception in write() method, deactivating: Traceback (most recent call last):
      File "bluetooth_repl.py", line 74, in write
      File "bluetooth_repl.py", line 29, in schedule_in
    OSError: (-258, 'ESP_ERR_INVALID_ARG')

this traces back to:
        _timer.init(mode=machine.Timer.ONE_SHOT, period=delay_ms, callback=_wrap)

I have tried creating timers from repl for the ESP32C3, and always get the error above on calling init, regardless of the subset of arguments provided or the values used. For reference:

>>> from machine import Timer
>>> t1=Timer(-1)
>>> t1
Timer(3fccd9c0; alarm_en=1073741823, auto_reload=1070351008, counter_en=1070350504)
>>> t1.init(mode=Timer.ONE_SHOT, period=10, callback=p)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-258, 'ESP_ERR_INVALID_ARG')

I realize C3 support is a work in progress, and am delighted overall, tnx! Perhaps the timers aren't fully supported yet on this device?

tnx

bdambrosio, 2021-10-02:
btw, above behavior is verified on both 1.17 stable and 1.17 20211002 unstable. tnx

cnadler86, 2024-06-28:
I can just tell from my experience that esp32c3 has only 2 HW tiners and the IDs are 0 and 2.
Maybe this should be put into paper in the [documentation](https://docs.micropython.org/en/latest/esp32/quickref.html)