micropython/micropython #7872: digital high level dectection is not working correctly

CxZMoE, 2021-10-03:
I'm previously using micropython 1.14, and read on digital pin can return high when the input voltage is VCC/2.
When I upgrated from 1.14 to 1.17, the input voltage need to be much higher to trigger high signal.
What is going on,and what can I do to fix it?

robert-hh, 2021-10-03:
According to the datasheet, the dependable value of a GPIO high level is > 0.75 * Vdd, which is >~2.5V at an Vdd of 3.3V. Anything below is not dependable. The low level is defined as < 0.25 Vdd, or < ~0.8 V. The range between 0.25 * Vdd and 0.75 * Vdd is unspecified. It may work, it may not, it will vary with temperature and by GPIO pin an device. This is a hardware property. I do not see any feature in the technical manuals (like hysteresis) to change it. What exactly is the required high level now, and did that change for the same device or a different device?
b.t.w.

dpgeorge, 2022-03-30:
Closing due to inactivity.

closed 2022-03-30 by dpgeorge