micropython/micropython #7879: compile faild: w5500 ethernet module for black_f407ve(stm32)

JeterC, 2021-10-04:
I have successfully compiled the w5500 network module for pyboard through the following command:
>make MICROPY_PY_WIZNET5K=5500

But when I try to use another development board, it fails when compiling
>make MICROPY_PY_WIZNET5K=5500 BOARD=BLACK_F407VE

The results are as follows:
>root@bananapim2ultra:/mnt/sda/micropython/micropython-Version1.7-ma
>ster/ports/stm32# make MICROPY_PY_WIZNET5K=5500 BOARD=BLACK_F407VE
>Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
>GEN build-BLACK_F407VE/genhdr/qstrdefs.collected.h
>QSTR not updated
>GEN build-BLACK_F407VE/genhdr/compressed.collected
>Compressed data not updated
>CC mpnetworkport.c
>mpnetworkport.c: In function'pyb_lwip_poll':
>mpnetworkport.c:58:5: error: implicit declaration of function'wiznet5k_poll' [-Werror=implicit-function-declaration]
>    58 | wiznet5k_poll();
>       | ^~~~~~~~~~~~~
>cc1: all warnings being treated as errors
>make: *** [../../py/mkrules.mk:77: build-BLACK_F407VE/mpnetworkport.o] Error 1 `

jonnor, 2024-10-01:
Did you manage to resolve the problem? Is it still a problem in recent MicroPython, version 1.23 or later?

jonnor, 2025-03-03:
Missing information to reproduce.

closed 2025-03-03 by jonnor