micropython/micropython #7881: uhashlib module object has no attribute sha1

ernesto75, 2021-10-05:
When using the micropython port for rp2 board, and trying to use uhashlib.sha1(), it gives error: module object has no attribute sha1. Only sha256 is implemented. Please add sha1 in uhashlib as well. I also tried to compile micropython from source, editing the file /micropython/ports/rp2/mpconfigport.h  By default only #define MICROPY_PY_UHASHLIB (1) is present; i tried to add #define MICROPY_PY_UHASHLIB_SHA1 (1) but the compiling process results in this error:
[  4%] Building C object CMakeFiles/firmware.dir/nfsshare/pico/micropython/extmod/moduhashlib.c.obj
In file included from /pico/micropython/py/mpstate.h:35:0,
                 from /pico/micropython/py/runtime.h:29,
                 from /pico/micropython/extmod/moduhashlib.c:30:
/pico/micropython/extmod/moduhashlib.c:249:60: error: 'uhashlib_sha1_digest' undeclared here (not in a function); did you mean 'uhashlib_sha256_digest'?
 STATIC MP_DEFINE_CONST_FUN_OBJ_1(uhashlib_sha1_digest_obj, uhashlib_sha1_digest);
                                                            ^
/pico/micropython/py/obj.h:337:42: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_1'
     {{&mp_type_fun_builtin_1}, .fun._1 = fun_name}
                                          ^~~~~~~~
/pico/micropython/extmod/moduhashlib.c:260:17: error: 'uhashlib_sha1_make_new' undeclared here (not in a function); did you mean 'uhashlib_sha256_make_new'?
     .make_new = uhashlib_sha1_make_new,
                 ^~~~~~~~~~~~~~~~~~~~~~
                 uhashlib_sha256_make_new
/pico/micropython/extmod/moduhashlib.c:171:17: error: 'uhashlib_sha1_update' used but never defined [-Werror]
 STATIC mp_obj_t uhashlib_sha1_update(mp_obj_t self_in, mp_obj_t arg);
                 ^~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[3]: *** [CMakeFiles/firmware.dir/build.make:2219: CMakeFiles/firmware.dir/pico/micropython/extmod/moduhashlib.c.obj] Error 1

NikhitaR-IFX, 2023-02-09:
Hi @ernesto75,
Did you manage any workaround here? I am facing same issue for a port in development while enabling the network module. Any ideas?

jimmo, 2023-02-15:
If you need sha1 then you can install it from micropython-lib.

ernesto75, 2023-02-16:
> Hi @ernesto75, Did you manage any workaround here? I am facing same issue for a port in development while enabling the network module. Any ideas?

No ideas, also, afaik, there's no sha1 in https://github.com/micropython/micropython-lib/tree/master/python-stdlib/hashlib/hashlib

jimmo, 2023-02-17:
@ernesto75 My mistake, sorry about that. OK, yes, we will enable SHA1 on all rp2040 boards before the next release (very soon!).

dpgeorge, 2023-02-17:
`hashlib.sha1` is now enabled on all rp2-based boards; see c6140ad0d4add2742a9a7645e74a7d2a3affabe7

closed 2023-02-17 by dpgeorge

NikhitaR-IFX, 2023-02-17:
Thanks @dpgeorge for confirming!