micropython/micropython #7884: CPython difference for __contains__ method

stinos, 2021-10-06:
```
class TestContains:
  def __contains__(self, arg):
    return arg

print(0 in TestContains())
print(1 in TestContains())
print(2 in TestContains())
print(0 not in TestContains())
print(1 not in TestContains())
print(2 not in TestContains())
```

CPython:
```
False
True
False
True
False
False
```

because https://docs.python.org/3/reference/expressions.html#membership-test-operations

> For user-defined classes which define the __contains__() method, x in y returns True if y.__contains__(x) returns a true value, and False otherwise.

MicroPython:
```
0
1

True
False
False
```

Unexpected results so definitely worthy of a cpydiff addition, or should this be fixed?

dlech, 2021-10-06:
Why does `print(2 in TestContains())` print `False` and an empty string? I would expect it to print `True` and `2`.

stinos, 2021-10-06:
> Why does `print(2 in TestContains())` print `False` and an empty string? I would expect it to print `True` and `2`.

Sorry, because I pasted a previous version of the test code :) Fixed.

dpgeorge, 2022-03-30:
I think this should be fixed, because it can lead to incorrect logic.

dpgeorge, 2022-04-20:
Fixed by 1ded8a2977a84777f339ea7e7c788b11d75a6cce

closed 2022-04-20 by dpgeorge