micropython/micropython #7887: esp32 latest builds PWM bug

Carglglz, 2021-10-09:
In the latest builds (esp32-20210924-unstable-v1.17-68-g35fb90bd5.bin, esp32-20211009-unstable-v1.17-74-gd42cba0d2.bin) PWM object fails to set frequency and duty cycle.

Probably there is a mismatch between the build process and PR #7817

PWM methods:
```
['__class__', 'deinit', 'duty', 'freq', 'init']
```
So following latest docs `duty_ns` and `duty_u16` are missing

And the errors are:

```
>>> pwm.duty(200)
E (227090) ledc: ledc_set_duty(485): speed_mode argument is invalid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-258, 'ESP_ERR_INVALID_ARG')

>>> pwm.freq(200)
E (256550) ledc: ledc_bind_channel_timer(220): speed_mode argument is invalid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: failed to bind timer to channel

```

robert-hh, 2021-10-09:
"So following latest docs duty_ns and duty_u16 are missing"
It's common in MicroPython that not every port offers every method, even if it would be easy to support them here as a scaling of the arguments.
About the error message: I cannot replicate them, using e.g. Pin 14 for the PWM output. Which Pin did you try?

robert-hh, 2021-10-09:
The only thing that the docs say about deinit() is, that it stops the PWM output. with v1.17 the PWM changed a lot, from a more static single frequency design to dynamic multiple frequencies. Still deinit() stops the output, but it seems to release resources bound to that python object. So what seems to be missing is a stop()/start() pair of methods, unless deinit()/init() can be considered as such.

robert-hh, 2021-10-11:
What seems undefined is the numeric range of duty(). For esp32, it is implemented as 0..1023. I implemented it for MIMXRT as percent value. ESP8266 has the 0..1023 range, WiPy 0..10000, pyb has a percent setting in the pyb module.
duty_ns() and duty_u16() is well defined.

kdschlosser, 2021-10-14:
I am going to chime in here. With the current code the duty resolution is capped for some reason.

This is a version that i am running where the  duty resolution can be changed by the user so it will support as large or as small a resolution as needed. I also added a function that will return the maximum duty that can be set.

I know it will need to be refined and maybe have the resolution values set as class constants. It will give the general idea.

<details>
  <summary>ports\esp32\macine_pwm.c</summary>
  
 ```c
/*
 * This file is part of the Micro Python project, http://micropython.org/
 *
 * The MIT License (MIT)
 *
 * Copyright (c) 2016-2021 Damien P. George
 * Copyright (c) 2018 Alan Dragomirecky
 * Copyright (c) 2020 Antoine Aubert
 * Copyright (c) 2021 Ihor Nehrutsa
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
 */

#include "py/runtime.h"
#include "py/mphal.h"

#include "driver/ledc.h"
#include "esp_err.h"

#include <math.h>

#define PWM_DBG(...)
// #define PWM_DBG(...) mp_printf(&mp_plat_print, __VA_ARGS__)

// Total number of channels
#define PWM_CHANNEL_MAX (LEDC_SPEED_MODE_MAX * LEDC_CHANNEL_MAX)
typedef struct _chan_t {
    // Which channel has which GPIO pin assigned?
    // (-1 if not assigned)
    gpio_num_t pin;
    // Which channel has which timer assigned?
    // (-1 if not assigned)
    int timer_idx;
} chan_t;
// List of PWM channels
STATIC chan_t chans[PWM_CHANNEL_MAX];

// channel_idx is an index (end-to-end sequential numbering) for all channels
// available on the chip and described in chans[]
#define CHANNEL_IDX(mode, channel) (mode * LEDC_CHANNEL_MAX + channel)
#define CHANNEL_IDX_TO_MODE(channel_idx) (channel_idx / LEDC_CHANNEL_MAX)
#define CHANNEL_IDX_TO_CHANNEL(channel_idx) (channel_idx % LEDC_CHANNEL_MAX)

// Total number of timers
[... 622 lines cut ...]
// This PWM class is implemented by machine_pwm.c.
extern const mp_obj_type_t machine_pwm_type;

// A port must provide implementations of these low-level PWM functions, either as global
// linker symbols, or included directly if MICROPY_PY_MACHINE_PWM_INCLUDEFILE is defined.
#ifndef MICROPY_PY_MACHINE_PWM_INCLUDEFILE
void mp_machine_pwm_print(const mp_print_t *print, mp_obj_t self_in, mp_print_kind_t kind);
mp_obj_t mp_machine_pwm_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *args);
void mp_machine_pwm_init_helper(machine_pwm_obj_t *self, size_t n_args, const mp_obj_t *pos_args, mp_map_t *kw_args);
void mp_machine_pwm_deinit(machine_pwm_obj_t *self);
mp_obj_t mp_machine_pwm_maximum_duty_get(machine_pwm_obj_t *self)
mp_obj_t mp_machine_pwm_freq_get(machine_pwm_obj_t *self);
void mp_machine_pwm_freq_set(size_t n_args, const mp_obj_t *pos_args, mp_map_t *kw_args);
mp_obj_t mp_machine_pwm_duty_get(machine_pwm_obj_t *self);
void mp_machine_pwm_duty_set(machine_pwm_obj_t *self, mp_int_t duty);
mp_obj_t mp_machine_pwm_duty_get_u16(machine_pwm_obj_t *self);
void mp_machine_pwm_duty_set_u16(machine_pwm_obj_t *self, mp_int_t duty_u16);
mp_obj_t mp_machine_pwm_duty_get_ns(machine_pwm_obj_t *self);
void mp_machine_pwm_duty_set_ns(machine_pwm_obj_t *self, mp_int_t duty_ns);
#endif

#endif // MICROPY_INCLUDED_EXTMOD_MACHINE_PWM_H
```

</details>

IhorNehrutsa, 2021-12-03:
May closed. Fixed in #7907

closed 2021-12-03 by Carglglz