micropython/micropython #7891: Having problem declaring default  arguement values

kdschlosser, 2021-10-10:
This might be considered a question or it could be a bug, not sure which I have tried every way I can think of to set a default value for a function argument as a float.

Here is the code block for what I am working on

```c
STATIC mp_obj_t machine_hw_can_get_timings(size_t n_args, const mp_obj_t *pos_args, mp_map_t *kw_args) {
    enum {
        ARG_bitrate,
        ARG_bitrate_tolerance,
        ARG_sample_point,
        ARG_sample_point_tolerance,
        ARG_bus_length,
        ARG_transceiver_delay
        //ARG_self
    };

    static const mp_arg_t allowed_args[] = {
        {MP_QSTR_bitrate, MP_ARG_REQUIRED | MP_ARG_INT, {.u_int = 500000}},
        {MP_QSTR_bitrate_tolerance, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = 2.0 }},
        {ARG_sample_point, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = 0.0 }},
        {MP_QSTR_sample_point_tolerance, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = 5.0 }},
        {MP_QSTR_bus_length, MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = 1}},
        {MP_QSTR_transceiver_delay, MP_ARG_KW_ONLY | MP_ARG_INT, {.u_int = 150}}

    };

    mp_arg_val_t args[MP_ARRAY_SIZE(allowed_args)];
    mp_arg_parse_all(n_args, pos_args, kw_args, MP_ARRAY_SIZE(allowed_args), allowed_args, args);

    uint32_t bitrate = args[ARG_bitrate].u_int;
    uint16_t bitrate_tolerance = (uint16_t) (mp_obj_get_float_to_f(args[ARG_bitrate_tolerance].u_obj) * 100.0);
    uint16_t sample_point = (uint16_t) (mp_obj_get_float_to_f(args[ARG_sample_point].u_obj) * 100.0);
    uint16_t sample_point_tolerance = (uint16_t) (mp_obj_get_float_to_f(args[ARG_sample_point_tolerance].u_obj) * 100.0);
    uint16_t bus_length = args[ARG_bus_length].u_int;
    uint16_t transceiver_delay = args[ARG_transceiver_delay].u_int;
```

I know that due to memory concerns it is not ideal to use floats and I could use integers but I think it would make it more confusing to use as I need to have the passed values be fractional.

Is there a way to set the parameter defaults as a float?

kdschlosser, 2021-10-10:
Thanks for the help with this. have been trying all kinds of things over the last 2 days. I should have asked 2 days ago!! LOL.

I am going to give it a go and see if it compiles and runs properly.

kdschlosser, 2021-10-10:
doesn't work.

```
../machine_can.c: In function 'machine_hw_can_get_timings':
../machine_can.c:489:18: error: unknown type name 'mp_obj_float_t'
     static const mp_obj_float_t default_bitrate_tolerance_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(2.0)};
                  ^~~~~~~~~~~~~~
../machine_can.c:489:5: warning: braces around scalar initializer
     static const mp_obj_float_t default_bitrate_tolerance_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(2.0)};
     ^~~~~~
../machine_can.c:489:5: note: (near initialization for 'default_bitrate_tolerance_obj')
../machine_can.c:489:67: warning: initialization of 'int' from 'const mp_obj_type_t *' {aka 'const struct _mp_obj_type_t *'} makes integer from pointer without a cast [-Wint-conversion]
     static const mp_obj_float_t default_bitrate_tolerance_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(2.0)};
                                                                   ^
../machine_can.c:489:67: note: (near initialization for 'default_bitrate_tolerance_obj')
In file included from /home/osboxes/esp/micropython/py/obj.h:31,
                 from ../machine_can.c:27:
../machine_can.c:489:104: warning: excess elements in scalar initializer
     static const mp_obj_float_t default_bitrate_tolerance_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(2.0)};
                                                                                                        ^~~
/home/osboxes/esp/micropython/py/mpconfig.h:716:32: note: in definition of macro 'MICROPY_FLOAT_CONST'
 #define MICROPY_FLOAT_CONST(x) x##F
                                ^
../machine_can.c:489:104: note: (near initialization for 'default_bitrate_tolerance_obj')
     static const mp_obj_float_t default_bitrate_tolerance_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(2.0)};
                                                                                                        ^~~
/home/osboxes/esp/micropython/py/mpconfig.h:716:32: note: in definition of macro 'MICROPY_FLOAT_CONST'
 #define MICROPY_FLOAT_CONST(x) x##F
                                ^
../machine_can.c:490:18: error: unknown type name 'mp_obj_float_t'
     static const mp_obj_float_t default_sample_point_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(0.0)};
                  ^~~~~~~~~~~~~~
../machine_can.c:490:5: warning: braces around scalar initializer
     static const mp_obj_float_t default_sample_point_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(0.0)};
     ^~~~~~
../machine_can.c:490:5: note: (near initialization for 'default_sample_point_obj')
../machine_can.c:490:62: warning: initialization of 'int' from 'const mp_obj_type_t *' {aka 'const struct _mp_obj_type_t *'} makes integer from pointer without a cast [-Wint-conversion]
     static const mp_obj_float_t default_sample_point_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(0.0)};
                                                              ^
../machine_can.c:490:62: note: (near initialization for 'default_sample_point_obj')
In file included from /home/osboxes/esp/micropython/py/obj.h:31,
                 from ../machine_can.c:27:
../machine_can.c:490:99: warning: excess elements in scalar initializer
     static const mp_obj_float_t default_sample_point_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(0.0)};
                                                                                                   ^~~
/home/osboxes/esp/micropython/py/mpconfig.h:716:32: note: in definition of macro 'MICROPY_FLOAT_CONST'
 #define MICROPY_FLOAT_CONST(x) x##F
                                ^
../machine_can.c:490:99: note: (near initialization for 'default_sample_point_obj')
     static const mp_obj_float_t default_sample_point_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(0.0)};
                                                                                                   ^~~
/home/osboxes/esp/micropython/py/mpconfig.h:716:32: note: in definition of macro 'MICROPY_FLOAT_CONST'
 #define MICROPY_FLOAT_CONST(x) x##F
                                ^
../machine_can.c:491:18: error: unknown type name 'mp_obj_float_t'
     static const mp_obj_float_t default_sample_point_tolerance_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(5.0)};
                  ^~~~~~~~~~~~~~
../machine_can.c:491:5: warning: braces around scalar initializer
     static const mp_obj_float_t default_sample_point_tolerance_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(5.0)};
     ^~~~~~
../machine_can.c:491:5: note: (near initialization for 'default_sample_point_tolerance_obj')
../machine_can.c:491:72: warning: initialization of 'int' from 'const mp_obj_type_t *' {aka 'const struct _mp_obj_type_t *'} makes integer from pointer without a cast [-Wint-conversion]
     static const mp_obj_float_t default_sample_point_tolerance_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(5.0)};
                                                                        ^
../machine_can.c:491:72: note: (near initialization for 'default_sample_point_tolerance_obj')
In file included from /home/osboxes/esp/micropython/py/obj.h:31,
                 from ../machine_can.c:27:
../machine_can.c:491:109: warning: excess elements in scalar initializer
     static const mp_obj_float_t default_sample_point_tolerance_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(5.0)};
                                                                                                             ^~~
[... 99 lines cut ...]
[... cut ...]
                                                                   ^~~~~~~~~~
/home/osboxes/esp/micropython/py/obj.h:309:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define MP_ROM_PTR(p) (p)
                       ^
../machine_can.c:580:81: note: in expansion of macro 'MP_ROM_PTR'
         {MP_QSTR_sample_point_tolerance, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = MP_ROM_PTR(&default_sample_point_tolerance_obj) }},
                                                                                 ^~~~~~~~~~
```

dlech, 2021-10-10:
Because it is only declared in a header file, not defined. If you want a quick hack, you can copy the definition from `py/objfloat.c` and add it to your source code file. It just won't work with other object representations as described in the issue I linked.

```c
typedef struct _mp_obj_float_t {
    mp_obj_base_t base;
    mp_float_t value;
} mp_obj_float_t;
```

kdschlosser, 2021-10-11:
```
home/osboxes/esp/micropython/py/obj.h:309:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define MP_ROM_PTR(p) (p)
                       ^
../machine_can.c:500:67: note: in expansion of macro 'MP_ROM_PTR'
         {ARG_sample_point, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_obj = MP_ROM_PTR(&default_sample_point_obj) }},

```

OK i declared the struct in my code and that did work. However I do have the above being shown. That's probably because there is no MP_ROM_PTR macro that handles mp_obj_float_t. so I removed the use of the macro and I am compiling now. we will see what happens.

kdschlosser, 2021-10-11:
I was right. 
```c
(void *)&default_tolerance_obj
```

fixes the problem. well it compiles I still have to test it.