micropython/micropython #7892: MicroPython CAN API

kdschlosser, 2021-10-11:
There has been a request to create a CAN API for adding CAN support for MCU's. The STM32 and the ESP32  are the 2 MCU's that have CAN support. I have become pretty familiar with the ESP32 side of things and if there is someone that is willing to spear head the STM32 side of things we can get CAN support added to MicroPython for the ESP32.

tannewt, 2021-10-18:
Here is the CircuitPython `canio` API doc for reference: https://circuitpython.readthedocs.io/en/7.0.x/shared-bindings/canio/index.html

kdschlosser, 2021-10-19:
@tannewt 

Thanks for the link. I went poking about the CircuitPython code and checked out what they have done.

Because of the large number of differences in how an MCU has it's API for a CAN interface I think it would be best to have a a CAN binding be it's own module like what is done in CircuitPython. The one thing I think that should be done is an import error should get raised for MCUs that does not support CAN. The module should exist and only the import error gets raised with a message of "Not supported by x MCU" where x is the MCU model. This way at least there is an informative message as to why the import error has occurred. A user would then be able to react on that import error so their software would be able to operate differently if needs be.

dabbede, 2021-11-08:
> It's not built into the TWAI interface. I have brought this up with the folks over at Expressif and I was told that it would be added to the next major release

[ESP32 documentation](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) state that TWAI peripheral support interrupt handling, and in particular it has support for rx and tx interrupts. Please look at section 21.5.3 Interrupt Management.
However, I'm not familiar with uPython and I don't know how to put it in code. In ESP-IDF you have to deal with the intr_flags of the peripheral config struct, [which ultimately sets the appropriate registers here](https://github.com/espressif/esp-idf/blob/d0dd9ce1b9825c29ae2b462556c4bf3965ab774b/components/driver/twai.c#L447).
Hope this helps

projectgus, 2023-08-30:
We've opened a new issue to collect the requirements and plan for implementing a cross-platform `machine.CAN` interface - please see #12337 . The discussion here is really helpful in framing the requirements, thanks!

I'm going to close this in favour of the linked issue.

closed 2023-08-30 by projectgus