micropython/micropython #7908: Documentation refers to missing sdist_upip.py

wnienhaus, 2021-10-17:
In the [Creating distribution packages](https://docs.micropython.org/en/latest/reference/packages.html#creating-distribution-packages) section of the documentation, there is a broken link to a file `sdist_upip.py` that no longer exists in the [micropython-lib](https://github.com/micropython/micropython-lib) repo.

The file was removed without reason/explanation from micropython-lib (see [commit](https://github.com/micropython/micropython-lib/commit/3a6ab0b)). Perhaps there is a new/different way to create upip distribution packages?

This file is needed for creating upip compatible distribution packages for publishing on PyPI (i.e. for the process the documentation is documenting).

The link should point to a valid copy of `sdist_upip.py`, or the documentation should describe the correct way of creating distribution packages.

n-elia, 2022-03-15:
Bump!

That link is still broken.

jonnor, 2024-10-01:
upip has been replaced by mip in MicroPython 1.20, https://github.com/micropython/micropython/releases/tag/v1.20.0
Closing as no longer relevant.

closed 2024-10-01 by jonnor