micropython/micropython #7960: Does micropython have FLASH power-down protection?

china-zhangdj, 2021-11-01:
My pyboard has a problem that the file system in the flash is damaged due to a sudden power failure during operation, how should I solve it?

dpgeorge, 2021-11-02:
You can switch to littlefs: https://docs.micropython.org/en/latest/reference/filesystem.html#littlefs

china-zhangdj, 2021-11-08:
How should I set up my STM32 board so that it supports littlefs by default instead of fatfs. If fatfs is supported by default, if you have to switch from fatfs to littlefs through code every time, there is still a high probability that the program will not run.

dpgeorge, 2021-11-12:
> How should I set up my STM32 board so that it supports littlefs by default instead of fatfs

If you format the partition as littlefs it will remain that way and boot into littlefs each time.  You only have to run the code once to change it.