micropython/micropython #7964: Stm32f767 Ethernet cannot work without DHCP server.

wangshujun-tj, 2021-11-03:
When Ethernet is directly connected to f767 and PC, the IP address cannot be obtained. At this time, after manually configuring the IP address, the MPY status is 3, ifconfig displays the configured static IP address ，but the Ping operation cannot be completed.
If static IP is not configured, the state remains at 2
If you connect through a router, can automatically obtain the IP address, and then configure the static IP to work normally

It is recommended to modify the configuration to static IP and use it directly instead of DHCP

import network,time
nic=network.LAN()
nic.active(True)
print("starting ethernet ",end="")

while(nic.isconnected()==False):
    time.sleep(0.5)
    print(nic.status())
    #nic.ifconfig(("192.168.99.105", "255.255.255.0", "192.168.99.1", "8.8.8.8"))

nic.ifconfig(("192.168.99.105", "255.255.255.0", "192.168.99.1", "8.8.8.8"))
    
print("ok!")

#check ip numbers etc.
print(nic.ifconfig())

Ardenuf123, 2021-11-03:
Direct connection requires a cross-over cable?

wangshujun-tj, 2021-11-04:
The cross network cable is not used because the current PC network card supports the automatic line sequence cross function
The problem disappeared after changing a network cable. It seems that there is a problem with the network cable

closed 2021-11-04 by wangshujun-tj