micropython/micropython #7966: FW: The firmware download page looks wonderful now.

robert-hh, 2021-11-03:
Quite an impressive select of boards. And easy to use. The json files need more tags, but that can be added easily now that the structure is set up.

jimmo, 2021-11-04:
Thank you @robert-hh !! Sorry it's been a long time coming, working away at it in small chunks this year.

Shout out to @mcauser and @UnexpectedMaker for providing board images and additional metadata!! Thanks!!

We will do more with the features & filtering at some stage. Many boards do not currently offer firmware but the plan (I think) is to turn on automatic building of every board for all ports.

closed 2021-11-04 by robert-hh

dpgeorge, 2021-11-05:
For automatic building of mimxrt, rp2 and samd ports see #7972.