micropython/micropython #7967: Uping did not receive a return on f767 version 1.17

wangshujun-tj, 2021-11-04:
Confirm that the Ping initiated from the PC is completely normal, but the Ping initiated from the f767 does not receive a response
I see that the uping module was released two years ago. During this period, many versions of MPY have been updated. Is there an updated version that can be replaced

https://gist.github.com/shawwwn/91cc8979e33e82af6d99ec34c38195fb

import network
import uPing as nwk

nic=network.LAN()
nic.activate(1)
while(nic.isconnected()==False):
    pass

#check ip numbers etc.
print(nic.ifconfig())
print(nic.config("mac"))

nwk.ping("192.168.0.1")
nwk.ping("micropython.org")

robert-hh, 2021-11-04:
I'm using uping a lot, and I do not see any problem with that code. I had a problem once with getting no responses while using ping. The only device that reponded was my local router. But the reason for that were wrong checksums in the ICMP header, which were not properly created by the Ethernet controller. The router seemed to ignore the checksum error. Changing LWIP config such that the checksum was created by LWIP solved the problem.

wangshujun-tj, 2021-11-04:
I now use esp32 for testing, and the phenomenon is the same
As you said, the router can reply, but the PC does not respond. Ping the remote device through the router can also respond
It seems that this is the problem of the underlying protocol stack
The direct difference in packet capturing is that the data segment sends 24 bytes more
[image]
[image]

robert-hh, 2021-11-04:
The length of the data segment does not matter. You can change that with uping with the call option size=nn. The default is 64, of which the first 16 bytes are filled with ICMP request data. Can you 'ping' the PC from other devices?

wangshujun-tj, 2021-11-06:
There is no problem with uping in the environment of esp32. The initial error is that windows has set up a strict firewall
The test on f767 still failed

wangshujun-tj, 2021-11-08:
After replacing the MCU with h743zi, the test is normal. It is uncertain whether it is the firmware version problem or the hardware difference.
However, you can confirm that uping.py is correct