micropython/micropython #7969: Erroneous version check in main/CMakeList.txt?

magnusvw, 2021-11-04:
Hi!

I'm using esp-idf v4.4-dev-3573-gfb24a2941c

In esp-idf/tools/cmake/version.cmake:

`set(IDF_VERSION_MAJOR 5)
set(IDF_VERSION_MINOR 0)
set(IDF_VERSION_PATCH 0)

set(ENV{IDF_VERSION} "${IDF_VERSION_MAJOR}.${IDF_VERSION_MINOR}.${IDF_VERSION_PATCH}")`

In micropython/ports/esp32/main/CMakeLists.txt:

`if(IDF_VERSION_MINOR GREATER_EQUAL 1)
    list(APPEND IDF_COMPONENTS esp_netif)
endif()

if(IDF_VERSION_MINOR GREATER_EQUAL 2)
    list(APPEND IDF_COMPONENTS esp_system)
    list(APPEND IDF_COMPONENTS esp_timer)
endif()

if(IDF_VERSION_MINOR GREATER_EQUAL 3)
    list(APPEND IDF_COMPONENTS esp_hw_support)
    list(APPEND IDF_COMPONENTS esp_pm)
    list(APPEND IDF_COMPONENTS hal)
endif()`

This breaks the build for me, since these files never get included. I have "fixed" it by just commenting out the if clauses, but that is no proper solution. I know too little about the source code to make a proper fix. Anyone got any pointers, or know how to properly fix this into something that could get pushed into the repository?

magnusvw, 2021-11-04:
https://github.com/micropython/micropython/pull/7970

UnexpectedMaker, 2021-11-06:
You are not using 4.4dev. You are using 5 now if you have latest master. 4.4 is now in its own release. You need to check that out instead of master for IDF.

magnusvw, 2021-11-07:
```
magnus@loki ~ $ which idf.py
/home/magnus/esp/esp-idf/tools/idf.py
magnus@loki ~ $ idf.py --version
Setting IDF_PATH environment variable: /home/magnus/esp/esp-idf
The following Python requirements are not satisfied:
pyparsing>=2.0.3,<2.4.0
idf-component-manager>=0.2.99-beta
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
python-socketio<5
kconfiglib==13.7.1
construct==2.10.54
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
Diagnostic information:
    IDF_PYTHON_ENV_PATH: (not set)
    Python interpreter used: /usr/bin/python
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: /home/magnus/bin:/home/magnus/.local/bin:/home/magnus/.rvm/gems/ruby-2.7.2/bin:/home/magnus/.rvm/gems/ruby-2.7.2@global/bin:/home/magnus/.rvm/rubies/ruby-2.7.2/bin:/home/magnus/.nvm/versions/node/v14.15.5/bin:/home/magnus/bin:/home/magnus/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/magnus/.dotnet:/home/magnus/.dotnet/tools:/home/magnus/esp/esp-idf/tools:/home/magnus/.rvm/bin:/home/magnus/.rvm/bin:/home/magnus/.dotnet:/home/magnus/.dotnet/tools:/home/magnus/esp/esp-idf/tools:/home/magnus/.rvm/bin

ESP-IDF v4.4-dev-3573-gfb24a2941c
magnus@loki ~ $ cat ~/esp/esp-idf/tools/cmake/version.cmake
set(IDF_VERSION_MAJOR 5)
set(IDF_VERSION_MINOR 0)
set(IDF_VERSION_PATCH 0)

set(ENV{IDF_VERSION} "${IDF_VERSION_MAJOR}.${IDF_VERSION_MINOR}.${IDF_VERSION_PATCH}")
```

UnexpectedMaker, 2021-11-07:
Latest master is a mess... switch to the 4.4 release branch
```
❯ idf.py --version                                                             
ESP-IDF v4.4-dev-3569-g6a7d83af19

❯ cat ~/esp/esp-idf/tools/cmake/version.cmake                                                                               
set(IDF_VERSION_MAJOR 4)
set(IDF_VERSION_MINOR 4)
set(IDF_VERSION_PATCH 0)

set(ENV{IDF_VERSION} "${IDF_VERSION_MAJOR}.${IDF_VERSION_MINOR}.${IDF_VERSION_PATCH}")

 ❯ git status
On branch IDF_4.4
Your branch is up to date with 'origin/IDF_4.4'.

```

UnexpectedMaker, 2021-11-07:
Also, this an IDF issue, not a MP one. Maybe create an issue on the IDF GitHub?

Current IDF5 is unstable and crashes MicroPython, so you should not be using it anyway, and should only be using 4.3.x or 4.4.

UnexpectedMaker, 2021-11-07:
Latest IDF master shows:
``` 
~/e/esp-idf >> master ⁝ ? >
❯ idf.py --version                                                                                          
ESP-IDF v5.0-dev-178-g417ef24b06
```
@magnusvw please close this issue as it's not correct and not a MP anyway, thanks.

magnusvw, 2021-11-07:
What is the opinion of the micropython team on this issue?

karfas, 2021-11-16:
@UnexpectedMaker: In the moment, IDF has no "v4.4" tag.
I use 4.4_dev ...

UnexpectedMaker, 2021-11-16:
> @UnexpectedMaker: In the moment, IDF has no "v4.4" tag.
> I use 4.4_dev ...

It's in it's own release branch.... `release/v4.4`

jonnor, 2025-10-03:
MicroPython now uses primarily IDF 5.x

closed 2025-10-03 by jonnor