micropython/micropython #7971: BLE CUSTOM MAC ADDRESS

toxer, 2021-11-04:
Can you externalize a funcrion for setup a cusom mac address for bluetooth device? There is a random mac address generator function. Can you update this function introducing a optional parameter for a fix mac address choose by user?

thx

jimmo, 2021-11-06:
Which port are you using?

On stm32 this is already supported for some configurations using mp_hal_get_mac, which you can override in your board definition.

But I agree this would be a useful general-purpose feature to set a custom public or random-static address, via ble.config(mac=) or something.

jimmo, 2021-11-06:
Also, out of curiosity, do you want to set a public address (i.e. from a purchased block) or a random static?

toxer, 2021-11-07:
Hi, I want to set custom mac address because I have an application that use
mac of a nearest device . Mac address must be fixed. I have found a
solution. I push it when I finish to test it.

jie-smile, 2022-09-09:
> Hi, I want to set custom mac address because I have an application that use mac of a nearest device . Mac address must be fixed. I have found a solution. I push it when I finish to test it. Il sab 6 nov 2021, 10:12 Jim Mussared ***@***.***> ha scritto:
> […](#)

Is this function available？

jimmo, 2022-09-09:
> Is this function available?

@jie-smile See above, do you want a public or random static address? Which port are you using?

jie-smile, 2022-09-13:
> > Is this function available?
> 
> @jie-smile See above, do you want a public or random static address? Which port are you using?

I want set a random static address in esp32

amplatzer, 2023-12-08:
> > > Is this function available?
> > 
> > 
> > @jie-smile See above, do you want a public or random static address? Which port are you using?
> 
> I want set a random static address in esp32

Have you found the solution? also need this function.