micropython/micropython #7981: rp2: Memory error in thread allocation

jimmo, 2021-11-09:
Saw this on stackoverflow: https://stackoverflow.com/questions/69796458/is-there-a-memory-leak-in-micropython-hex-function

Here's a simple repro (that also shows the issue isn't with `hex`). If you run it with a smaller number of iterations (i.e. 0x1000) or not in a thread, it successfully runs to completion and prints `"finished"`.

```py
import _thread

def test():
    for i in range(0x1800):
        bytes(0)
    print("finished")

_thread.start_new_thread(test, ())
```

Note that it does successfully trigger an automatic collection, then fails about 144 allocations later. (Add `print(gc.mem_free())` before the `bytes(0)` to see this).

peterhinch, 2021-11-09:
See also [this forum post](https://forum.micropython.org/viewtopic.php?f=21&t=11410).

manpowre, 2021-11-12:
> 
> 
> See also [this forum post](https://forum.micropython.org/viewtopic.php?f=21&t=11410).

Its not the same issue, I tried this code here with gc.threshold(5423) and lower, and it still crashes.

dpgeorge, 2022-03-01:
Fixed by 2d47020e15b82b7f021e0b52dabe75baf5cb4c4d

closed 2022-03-01 by dpgeorge