micropython/micropython #7982: esp32: No OTA images generated anymore since switch to cmake

mirko, 2021-11-10:
For versions using an ESP IDF SDK version using gmake I used to find binary files for (almost) each partition inside the build dir (`ports/esp32/build-$$BOARD$$/`):
 - bootloader.bin
 - partitions.bin
 - application.bin
 - firmware.bin
It seems, though, that at least the OTA image (formerly `application.bin`) is missing since switching to cmake.

Given that at least the upstream supported board `SIL_WESP32` also uses OTA (`CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions-16MiB-ota.csv` in sdkconfig), I would have expected to also find an OTA binary image (or an equivalent) - however I can't find anything alike.

The only thing I can see is the file `ports/esp32/build-$$BOARD$$/ota_data_initial.bin`, which however is basically empty (well, 8KiB of 0xFF).

What's the supposedly correct way generating/obtaining OTA images with micropython v1.17 onwards?

dpgeorge, 2021-12-01:
If you look at how `makeimg.py` is invoked from `Makefile` it is still basically the same as it was pre-cmake and shows which .bin files make up the final `firmware.bin`:
```
        @$(PYTHON) makeimg.py \
                $(BUILD)/sdkconfig \
                $(BUILD)/bootloader/bootloader.bin \
                $(BUILD)/partition_table/partition-table.bin \
                $(BUILD)/micropython.bin \
                $(BUILD)/firmware.bin
```
(sdkconfig is passed in so `makeimg.py` can extract some config values)

So `application.bin` is now `micropython.bin`; this is set by the line `project(micropython)` at the end of `CMakeLists.txt`.

Does that help?

mirko, 2021-12-01:
I think it does - thanks a lot. Will close once tested.