micropython/micropython #7994: rp2: sidestep example?

oe8bck, 2021-11-15:
Hi,

I am quite new to PIO concept, I would like to make a simplex communication on one pin. To be more precise the pin concept is open-drain (external pull-up) and the communication is triggered by the Pico by pulling the line LOW. This is working in my PIO program, but after this pulse, the pin has to be set high ohmic in order that the peripheral can answer by pulling the wire LOW. If I look into the datashee, I think this is possible with a sidestep.

Does anybody have an example switching the pin mode via sidestep in Python? Alternatively I would also have a look on an example just switching the output pin to high ohmic input...

miketeachman, 2021-11-15:
Do you know about the [Micropython Forum](https://forum.micropython.org/)?  It's the best place to ask for technical help.  My suggestion is to close this Issue and seek help on the Forum.

dpgeorge, 2021-11-16:
There are examples in https://github.com/micropython/micropython/tree/master/examples/rp2

And some documentation here: http://docs.micropython.org/en/latest/rp2/tutorial/pio.html and http://docs.micropython.org/en/latest/library/rp2.html#pio-assembly-language-instructions

closed 2021-11-16 by dpgeorge

oe8bck, 2021-11-16:
> Do you know about the [Micropython Forum](https://forum.micropython.org/)? It's the best place to ask for technical help. My suggestion is to close this Issue and seek help on the Forum.

Thanks for the hint, I did so. So I'll close this issue.