micropython/micropython #7996: rp2: USB VCP keyboard interrupt not working

hoihu, 2021-11-16:
I'm on a latest build as of 16.11.2021 on a pico board and USB VCP keyboard interrupt does not seem to work.

On the console I tried the following

```
>>> import micropython
>>> micropython.kbd_intr(3)
>>> while True: print("Hello")
```

however the while loop does not break when pressing CTRL+C

I then tried to enable `#define MICROPY_KBD_EXCEPTION                   (1)` but did not succeed either

dpgeorge, 2021-11-16:
Trying the above with the latest build, it works for me, it can break out of the while-true loop.

Note that running `micropython.kbd_intr(3)` at the REPL won't have much of an effect because the interrupt char is set and reset each time a REPL line is run.

hoihu, 2021-11-17:
Thanks for the fast feedback! 

Ok that's really weird.. when trying the above code using pyserial's miniterm:

```
py -3 -m serial.tools.miniterm
```` 

it's not possbile to interrupt. And something else is strange: If I disconnect the device from USB, the terminal completly freezes. Since we are using pyserial extensively with micropython, I doubt it's an issue with that tool.

However, If I use `putty`, it does seem to work though!

hoihu, 2021-11-17:
Also as a side note, the same happens on another PC, using linux.  So it doesn't seem to be related to the host OS.

hoihu, 2021-11-17:
Ok on further testing, even with `putty` it does not work (reliable): When pressing some keys *before* CTRL+C, putty does not break out of the loop either.

Apparently, putty sends `\n` and pyserial `\r\n` . So when issuing pyserial with:

``` 
py -3 -m serial.tools.miniterm --eol=cr
```

also pyserial seems to "work". But in both cases, the issue remains that if there are keys pressed before CTRL+C is pressed, it's not possible to bail out of the loop.

also modifying the test with a `time.sleep` call after the print did not work:  ```>> while True: print("Hello"); time.sleep(1) ```

hoihu, 2021-11-17:
Wireshark shows a problem.. here is the log output for the following 

```
>> while 1: print("HELLO"); time.sleep(1);
```

While this runs, I start entering a few characters, then CTRL+C (which is not working). I observed the following output:

[image]

The above shows that the "Hello" strings are correctly transmitted, however when starting to press "a", then "s" in the terminal, the "s" is not acknowledged anymore and any further keypresses are no more visible on wireshark. 

My assumption is that the Pico NACK's the endpoint for some reason.

hoihu, 2021-11-17:
So to summarize my understanding:

- pico seems to block the endpoint after receiving one character (if it isn't read by stdin.read)
- if the character is read (e.g. via sys.stdin.read()) it continues to receive more characters. This can be shown by modifying the test code with `>> while 1: print("HELLO"); sys.stdin.read()`  -> this way CTRL+C works always 
- this in turn has the consequence that CTRL+C does not work when a terminal is used with \r\n. But this is just one instantiation of the blocking.
- might be related to https://github.com/micropython/micropython/issues/6978

dpgeorge, 2021-11-18:
Yes I can confirm the problem, that USB serial blocks if the first incoming character is not processed.

hoihu, 2021-11-18:
Further digging, we were able to capture this USB traffic:

[image]

This is the CTRL+C (value 03) being NAK'd by the Pico. This frame is repeatet on the bus, since the host tries to send it the Pico. So bascially the USB is flooded with repeated tries to write the endpoint...

I'm not familiar with the tinyUSB stack. But digging around briefly, there seem to be a mechanism for stalling an endpoint when the ringbuffer is not able to digest the incoming data. But that can't be case here, since we only send a single character. However, maybe there is a bug there that this feauture is triggered somehow unintentionally

dpgeorge, 2021-11-18:
Would be interesting to investigate the following:
- does the mimxrt port have the same issue?
- does esp32s2/c3 have the same issue?
- what happens in CircuitPython, do they have the issue or work around it?

If this is expected TinyUSB behaviour (maybe it only has one OUT buffer and unless it gets read from it must continuously NAK the host) the way to fix it would be to always drain the incoming data to the stdin ringbuffer.

robert-hh, 2021-11-18:
Same problem on mimxrt. 
Using picocom@linux for the test, I can break the loop if Ctr-C is the first character. If it is not the first character, Ctrl-C does interrupt.

hoihu, 2021-11-18:
> If it is not the first character, Ctrl-C does interrupt.

did you mean "doesn't"?

robert-hh, 2021-11-18:
Indeed, changed.

zsquareplusc, 2021-11-18:
NAK on the USB level is a good way to hold back once the ring buffer is full. But that would still mean that there are cases where a CTRL+C is held on the PC's buffer and does not reach the device. One work around would be to discard the data (after looking for CTRL-C) when the ring buffer is full. An other option to support a `KeyboardInterrupt` at any time would be using BREAK, as this is transmitted on a different USB endpoint (just like RTS and DTR states, this is sent to endpoint 0/setup).

So using the receive callback from the USB stack to transfer to the ring buffer is certainly a good option.

However this is not the exact reason of the failure here. As it is already sending NAK after one or a few bytes and the ring buffer certainly would have more space. Though changing it to use the callback may solve this along the way...

dpgeorge, 2021-11-18:
I did another test where, while it's running a while-true loop, I send a bunch of characters at once over the USB serial port, with the characters ending in ctrl-C.  Eg send the string b"abc\x03".  This works, it breaks out of the loop.

So what I think is happening is that TinyUSB has room for exactly one OUT packet and sends NAK until that packet is processed on the device side.  That packet can contain 1 or up to (maybe) 64 characters.  In cases where you type the characters (slowly) into a terminal program, there's only one character per packet so typing "a" blocks everything because that uses up the one packet.

hoihu, 2021-11-20:
> So what I think is happening is that TinyUSB has room for exactly one OUT packet and sends NAK until that packet is processed on the device side. That packet can contain 1 or up to (maybe) 64 [...]

I agree, but I can't image this is done by design? So perhaps might be worth reporting upstream? 

Another approach  could be to consume the incoming data always (via the callback `tud_cdc_rx_cb`, line 451 cdc_device.c). But I imagine that would prevent the handshaking via `NACK`, so the USB stack would not know if input buffer is full. Which is somewhat a good feature to have.

A for the proposal of @zsquareplusc I think that could  be a neat addition to CTRL+C handling. I wonder if that could be implemented as an additional (SOS-) CTRL+C or something, in addition to standard CTRL+C handling so that special terminals would take advantage of that feature, whilst other remain backward compatible. It would solve issues where the input buffer is full (e.g. due to FW malfunctions etc) and the controller cannot be interrupted because USB NACK's the endpoint.

dhalbert, 2021-11-21:
Pinging @hathach for interest.

hoihu, 2021-11-21:
I did some further testing. This diff here implements the callback handling and adds a separate ringbuffer for usb:

```
diff --git a/ports/rp2/mphalport.c b/ports/rp2/mphalport.c
index 3385323db..e685ec42b 100644
--- a/ports/rp2/mphalport.c
+++ b/ports/rp2/mphalport.c
@@ -44,6 +44,19 @@ ringbuf_t stdin_ringbuf = { stdin_ringbuf_array, sizeof(stdin_ringbuf_array) };
 
 #endif
 
+#if MICROPY_HW_ENABLE_USBDEV
+
+STATIC uint8_t usb_ringbuf_array[256];
+ringbuf_t usb_ringbuf = { usb_ringbuf_array, sizeof(usb_ringbuf_array) };
+
+void tud_cdc_rx_cb(uint8_t itf) {
+    for (uint32_t bytes_avail = tud_cdc_n_available(itf); bytes_avail > 0; bytes_avail --) {
+        ringbuf_put(&usb_ringbuf, tud_cdc_read_char());
+    }
+}
+
+#endif
+
 #if MICROPY_KBD_EXCEPTION
 
 int mp_interrupt_char = -1;
@@ -70,7 +83,7 @@ uintptr_t mp_hal_stdio_poll(uintptr_t poll_flags) {
     }
     #endif
     #if MICROPY_HW_ENABLE_USBDEV
-    if (tud_cdc_connected() && tud_cdc_available()) {
+    if ((poll_flags & MP_STREAM_POLL_RD) && ringbuf_peek(&usb_ringbuf) != -1) {
         ret |= MP_STREAM_POLL_RD;
     }
     #endif
@@ -87,12 +100,9 @@ int mp_hal_stdin_rx_chr(void) {
         }
         #endif
         #if MICROPY_HW_ENABLE_USBDEV
-        if (tud_cdc_connected() && tud_cdc_available()) {
-            uint8_t buf[1];
-            uint32_t count = tud_cdc_read(buf, sizeof(buf));
-            if (count) {
-                return buf[0];
-            }
+        int c = ringbuf_get(&usb_ringbuf);
+        if (c != -1) {
+            return c;
         }
         #endif
         MICROPY_EVENT_POLL_HOOK
```

this seems to solve the issue. But as I see it, it's not ideal, since the NACK handshacking would no longer work. But at least the CTRL+C is solved.

dpgeorge, 2021-11-21:
> But as I see it, it's not ideal, since the NACK handshacking would no longer work.

To fix this, in `tud_cdc_rx_cb`, if the ringbuf is full then just return immediately.  That would miss ctrl-C if the ringbuf is full though....

dpgeorge, 2022-04-05:
Fixed by 587339022689187a1acbccc1d0e2425a67385ff7

closed 2022-04-05 by dpgeorge