micropython/micropython #8001: pyboard.py issue uploading 10MB file to 16MB ESP32-S2

TGiles1998, 2021-11-17:
Operating system
macOS Mojave

Python version
3.10

What Chip
ESP32 S2 16MB

Terminal Command and Output

pyboard.py --device /dev/cu.SLAB_USBtoUART -f cp model.tflite :
cp model.tflite :model.tflite
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/pyboard.py", line 582, in filesystem_command
    op(src, dest2)
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/pyboard.py", line 522, in fs_put
    self.exec_("w(" + repr(data) + ")")
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/pyboard.py", line 465, in exec_
    ret, ret_err = self.exec_raw(command, data_consumer=data_consumer)
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/pyboard.py", line 456, in exec_raw
    self.exec_raw_no_follow(command)
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/pyboard.py", line 434, in exec_raw_no_follow
    return self.raw_paste_write(command_bytes)
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/pyboard.py", line 399, in raw_paste_write
    raise PyboardError("unexpected read during raw paste: {}".format(data))
__main__.PyboardError: unexpected read during raw paste: b'e'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/pyboard.py", line 785, in <module>
    main()
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/pyboard.py", line 747, in main
    filesystem_command(pyb, args.files)
  File "/Library/Frameworks/Python.framework/Versions/3.10/bin/pyboard.py", line 598, in filesystem_command
    print(str(er.args[2], "ascii"))
IndexError: tuple index out of range

Hopefully there is a quick fix to this,

many thanks,

Thomas Giles

wangshujun-tj, 2021-11-30:
It is recommended to try to upload files from thorny's file view, or run FTP server on esp32 and use FTP to transfer files

TGiles1998, 2021-11-30:
Okay thank you very much!

closed 2021-12-01 by TGiles1998