micropython/micropython #8007: C++ version for Native machine code in .mpy files

shuaiye, 2021-11-18:
I built c version example, it works fine.
https://docs.micropython.org/en/latest/develop/natmod.html
but how Can I build a C++ version?
is it any example anywhere? or it hasn't be implemented yet.

jonnor, 2023-07-30:
Hi @shuaiye.
The bindings for MicroPython use the C ABI. So to use C++ the easiest would be to add C API to your code, and then use this in the .c files which are the MicroPython bindings.
Note that you may also need to add a C++ build step to your Makefile, and add the resulting .o files to `SRC_O`.

jimmo, 2023-08-03:
There is an example of writing a user module in `examples/usercmodule/cppexample` which covers the relevant steps for `extern "C"` etc that you could apply to a dynamic native module.

Off the top of my head, I would imagine this would work, however there would likely be some surprises using C++ as MicroPython's dynamic loader has several limitations that I can see affecting C++ programs: https://docs.micropython.org/en/latest/develop/natmod.html#supported-features-and-limitations

I'll close this issue as we don't have any plans right now to do any additional C++ specific support in dynamic native modules.

closed 2023-08-03 by jimmo