micropython/micropython #8009: Unable to compile natmod feature2 example

ShaunL36, 2021-11-18:
When I try to compile the features2 example, I get the following error:
```
GEN build/features2.config.h
CC main.c
main.c: In function 'add':
main.c:26:12: error: implicit declaration of function 'mp_obj_new_float'; did you mean 'mp_obj_new_list'? [-Werror=implicit-function-declaration]
     return mp_obj_new_float(mp_obj_get_float(x) + mp_obj_get_float(y));
            ^~~~~~~~~~~~~~~~
            mp_obj_new_list
main.c:26:29: error: implicit declaration of function 'mp_obj_get_float'; did you mean 'mp_obj_is_float'? [-Werror=implicit-function-declaration]
     return mp_obj_new_float(mp_obj_get_float(x) + mp_obj_get_float(y));
                             ^~~~~~~~~~~~~~~~
                             mp_obj_is_float
main.c:26:12: error: return makes pointer from integer without a cast [-Werror=int-conversion]
     return mp_obj_new_float(mp_obj_get_float(x) + mp_obj_get_float(y));
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [../../../py/dynruntime.mk:137: build/main.o] Error 1
```
I'm on Raspbian 10 (buster), there is the same error on Win10 and using Cygwin

This is my Makefile, I'm just change `ARCH` for `armv7m` because I want to run in Raspberry Pi Pico
```
# Location of top-level MicroPython directory
MPY_DIR = ../../..

# Name of module
MOD = features2

# Source files (.c or .py)
SRC = main.c prod.c test.py

# Architecture to build for (x86, x64, armv7m, xtensa, xtensawin)
ARCH = armv7m

# Include to get the rules for compiling and linking the module
include $(MPY_DIR)/py/dynruntime.mk
```

jimmo, 2021-11-18:
I think the idea is that native modules can only use float if there is hardware support (otherwise they would need to be using soft float, and my guess is that this would end up duplicating the support code). See https://github.com/micropython/micropython/blob/master/py/dynruntime.mk#L58 (if you set armv7m, then there's not float enabled).

FWIW, RP2040 is armv6m. See #6959 and #7616 (we need to do more to support armv6m).

jonnor, 2024-07-03:
Is this still a problem in latest git master?

ShaunL36, 2024-07-04:
Sorry, I am no longer responsible for this project, so I cannot verify it.

closed 2024-07-04 by ShaunL36