micropython/micropython #8011: Better support for OTA updates on esp32

bobveringa, 2021-11-18:
I work on an esp32 based device where we rely heavily on MicroPython for our development workflow and device code. We started using MicroPython around 2 years ago, and during that time we have developed a decent amount of MicroPython code. Part of that code is to support OTA (Over the Air) updates. Our current OTA Update system code was developed around 2 years ago for MicroPython v1.12 and is entirely written in python. 

We have recently identified our propriety code as somewhat of a risk, and we want to rewrite the code to use more native espressif functions. Ideally, we want to contribute this code back to the community as we also rely on their work for our platform to function. 

Is better support for OTA updates on esp32 something that should be added to MicroPython, and if so in what capacity?

The OTA update is specific to the firmware application, not files on the file system. We intend to develop a system that can download the update from both http and https sources.

I'm interested to hear what you think about this change.

bobveringa, 2022-02-03:
Is there any interest for this functionality?

I need to develop this functionality anyway. So, it would not require much more effort to integrated it into micropython.

jacklinquan, 2022-02-13:
Hi @bobveringa 
Thank you for your work on MicroPython ESP32 OTA!
I think many engineers who are trying to use MicroPython for production are interested in your contribute.
Like in my company, we sometimes verify our concept using MicroPython because it's quick and easy.
And then we reimplement the functionality in pure C code along with OTA, Flash Encryption and Secure Boot for production.
For large projects we probably have to do so due to the size of the code,
but for small projects(most of the cases) it would be much nicer to do everything just in MicroPython.

Is it also possible for you to share the python code you did for OTA?
I think many engineers are interested in it as well.

Many thanks again!

VarshneyNaupada, 2022-02-25:
Hi @bobveringa & @jacklinquan ,

I am interested to contribute.

Thank you !

bobveringa, 2022-02-25:
I had to make a deadline for an internal release, so I was unable to implement this feature in a user-module/ C module. Unfortunately, the python code is proprietary and utilizes some of our other internal modules, so even if the company would allow me to share it, it wouldn't be of any use.

I also use MicroPython at home for some projects, and it would be nice to have OTA support for those projects. So, I'll probably pick up this project when I have some spare time.

The mqboard project has an implementation of how you could do an OTA update. [This](https://github.com/tve/mqboard/blob/master/mqrepl/mqrepl.py#L36) file contains an OTA class. You could use this as an example. But I believe that method only works for updates over MQTT. If you need updates of HTTP then you would need to make some modifications.

AmirHmZz, 2022-03-13:
@bobveringa I've tried to write a `.bin` file to `ota_1` partition using that method, but It couldn't boot successfully. To be clear I'm writing `MicroPython` .bin file ( `[like this one](https://micropython.org/resources/firmware/esp32-ota-20220117-v1.18.bin)` ) to `ota_1` without modifying it. Am I doing something wrong ?

bobveringa, 2022-03-13:
@AmirHmZz I am not sure about the contents of that file, so I cannot say for certain. But as far as I am aware, all the files available for download are "firmware" binaries which include a bootloader.

What you need to write to the ota_1 partition is the application binary, which I do not think MicroPython provides. The application binary does not include the bootloader, so you should be able to boot from it. I cannot find a good source to back this up, but if you compile MicroPython from source you will see that there are 3 outputs, one of which is an application file.

Just writing the partition is not enough, you also need to set it as the boot partition. 

It should be noted that updating the application does not modify any data you have on the file system of the device. This is useful if you want to update your MicroPython version without removing code you already have on the device, but I think this is a very limited use case. If you want OTA support for your application, then you either need to update the file system, or you need to pre-compile your python code into a binary.

Pre-compiling your code does have benefits if you want to use your code in production (I'd say it is even a requirement for production) but if you are using this as a hobby, then I would avoid it, as it really changes your development workflow and makes everything a lot more complicated.