micropython/micropython #8015: Downloads: wrong file types for mimxrt build.

robert-hh, 2021-11-19:
The file types built for the mimxrt port are a little bit inconsistent. The Teensy boards require .hex files, the MIMXRT1xxx and Seed boards require .bin files. The files on the download page do not meet always and every day this rule. Is there any way to specify, which file type shall be created, like a tag in the json file?

dpgeorge, 2021-11-19:
> Is there any way to specify, which file type shall be created, like a tag in the json file?

The extensions are specified in `tools/autobuild/build-boards.sh`:
```
function build_mimxrt_boards {
    build_boards $1 $2 modmimxrt.c bin hex
}
```
Looking at the downloads page for Teensy https://micropython.org/download/TEENSY40/ it is inconsistent, and the sorting/order of releases seems wrong:
[image]

robert-hh, 2021-11-19:
That's what puzzled me. And for some m in mimxrt boards there is not .bin file at all.

dpgeorge, 2021-11-22:
Should be fixed.

closed 2021-11-22 by dpgeorge

robert-hh, 2021-11-22:
Thank you. All file types are visible now. The only slightly irritiating is, that for the mimxrt the .hex files are shown first, which is a little bit misleading, at least at the moment. @alphaFred will make a PR shortly with a generic bootloader for all mimxrt boards, which uses dfu-util for upload and accepts .hex files as well. That bootloader also does not erase the file system on firmware upload for Teensy boards. We are currently testing the bootloader to make sure, that it works well on all supported boards. The bootloader is bundled with the firmware, such that there is only one file type per device to take care of and chance are a little bit less to confuse users.

dpgeorge, 2021-11-22:
> he only slightly irritiating is, that for the mimxrt the .hex files are shown first, which is a little bit misleading, at least at the moment.

If that order needs changing, ie put .bin before .hex, then it requires a new entry in `board.json` which specifies the default extension (also possible the list of extensions that should be built and shown).

robert-hh, 2021-11-30:
> If that order needs changing, ie put .bin before .hex, then it requires a new entry in board.json which specifies the default extension (also possible the list of extensions that should be built and shown).

Does that entry already exist? Probably not.