micropython/micropython #8018: Provide images in non-DFU-friendly format (HEX, ELF)

krzema12, 2021-11-20:
I own STM32F429 board and it turns out that [DFU mode is not easily available there](https://github.com/micropython/micropython/wiki/Board-STM32F429). It's also challenging for me to build micropython from source to be able to deploy it from `make`.

From what I see, micropython images are provided only in DFU-friendly format. Did you consider publishing either HEX or ELF format? Then we could use the official STM32CubeProgrammer software to upload micropython.

Thanks,
Piotr

dpgeorge, 2021-11-21:
> Did you consider publishing either HEX or ELF format?

Yes it would be good to provide .hex as well as .dfu (since it seems that ST now actually prefer .hex files over .dfu with their tools..).

With the new `board.json` and auto build scripts this is now much easier.  See #8019.

dpgeorge, 2021-11-22:
Firmware is now provided in .dfu and .hex for all stm32 boards (downloads page may take a day to update though).

closed 2021-11-22 by dpgeorge

krzema12, 2021-11-22:
Thanks, that was so quick! I'll test it once the image is available.