micropython/micropython #8020: esp32 - uasyncio - anomaly

straga, 2021-11-21:
Create Task after pin irq. But Task created but not running. Running only after run run_wdt. If set await asyncio.sleep(10), all task from cb, run too after 10 sec. If set 1s run after 1s.

That normal?

10sec
[image]

1sec
[image]

0.01 sec
[image]

`import uasyncio as asyncio
import machine, _thread
import utime

async def run_wdt():
    print("PIN RUN")
    binary_pin()

    print("WDT RUN")
    while True:
        # print("WDT RESET")
        await asyncio.sleep(10)

def cb(binary_pin):
    print("Call Back: Create Task")
    period_in = utime.ticks_ms()
    loop = asyncio.get_event_loop()
    loop.create_task(binary_callback(binary_pin, period_in))

async def binary_callback(binary_pin, period_in):
    period_run = utime.ticks_ms()
    print(f"Run Task: after {period_run - period_in} ms")
    pin = binary_pin
    await asyncio.sleep(0.1)
    print(f"val: {pin.value()}")

def binary_pin():
    print("Binary Pin")
    p0 = machine.Pin(0, machine.Pin.IN, machine.Pin.PULL_UP)
    p0.irq(trigger=machine.Pin.IRQ_FALLING, handler=cb)

def main():

    # AsyncIO in thread
    loop = asyncio.get_event_loop()

    _ = _thread.stack_size(8 * 1024)
    _thread.start_new_thread(loop.run_forever, ())

    # Run Loader Task
    loop.create_task(run_wdt())

if __name__ == '__main__':
    print("MAIN")
    main()`

straga, 2022-01-29:
MicroPython v1.18-56-g517e82eb6 on 2022-01-29; WeAct Studio Core with STM32F411CE
If add fake_task work right. if comment  loop.create_task(fake_task()) work with anomaly.

```
import uasyncio as asyncio
import machine, _thread
import utime

pin_test = "PA0"

async def fake_task():
    while True:
        await asyncio.sleep(0.01)

async def run_wdt():
    print("PIN RUN")
    binary_pin()

    print("WDT RUN")
    while True:
        # print("WDT RESET")
        await asyncio.sleep(10)

def cb(binary_pin):
    print("Call Back: Create Task")
    period_in = utime.ticks_ms()
    loop = asyncio.get_event_loop()
    loop.create_task(binary_callback(binary_pin, period_in))

async def binary_callback(binary_pin, period_in):
    period_run = utime.ticks_ms()
    print(f"Run Task: after {period_run - period_in} ms")
    pin = binary_pin
    await asyncio.sleep(0.1)
    print(f"val: {pin.value()}")

def binary_pin():
    print("Binary Pin")
    p0 = machine.Pin(pin_test, machine.Pin.IN, machine.Pin.PULL_UP)
    p0.irq(trigger=machine.Pin.IRQ_FALLING, handler=cb)

def main():

    # AsyncIO in thread
    loop = asyncio.get_event_loop()

    _ = _thread.stack_size(8 * 1024)
    _thread.start_new_thread(loop.run_forever, ())

    # Run WDT Task
    loop.create_task(run_wdt())
    loop.create_task(fake_task())

if __name__ == '__main__':
    print("MAIN")
    main()
```

peterhinch, 2022-01-29:
@dpgeorge Is it valid to issue `create_task()` in a (soft) ISR context?

dpgeorge, 2022-01-30:
> Is it valid to issue `create_task()` in a (soft) ISR context?

No.  Because the soft ISR handler may preempt the inner asyncio loop.

It would be possible to use locks to forbid this (or raise errors if it happens), but that would probably decrease performance of asyncio and/or soft ISR calling.

Not sure what the best thing is to do here.  But `asyncio.ThreadSafeFlag` was added to help with communicating between ISRs and asyncio.

peterhinch, 2022-01-30:
Thanks, as I thought.

@straga 
I suggest you amend your code to use a `ThreadSafeFlag` with the ISR setting the flag. A separate task would loop, waiting on the flag before issuing `create_task()`. Note that in the current version of `uasyncio` you don't need to instantiate the event loop, you can issue `asyncio.create_task(foo())`. See [official docs](https://docs.micropython.org/en/latest/library/uasyncio.html) and [tutorial](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md).

straga, 2022-01-30:
Thanks a lot. Now works like a charm

```
import uasyncio as asyncio
import machine, _thread
import utime
from machine import Pin, Timer

pin_test = "PA0"

class AsyncPin:
    def __init__(self, pin, trigger):
        self.pin = pin
        self.flag = asyncio.ThreadSafeFlag()
        self.pin.irq(handler=lambda pin: self.flag.set(), trigger=trigger, hard=True)

    async def wait_edge(self):
        await self.flag.wait()
        return utime.ticks_ms()

async def binary_callback():
    pin_in = Pin(pin_test, Pin.IN, machine.Pin.PULL_UP)
    async_pin = AsyncPin(pin_in, Pin.IRQ_RISING)

    while True:
        t_push = await async_pin.wait_edge()
        print(f"Get: after {utime.ticks_ms() - t_push} ms")

async def run_wdt():
    print("WDT RUN")
    while True:
        # print("WDT RESET")
        await asyncio.sleep(10)

def main():

    # AsyncIO in thread
    loop = asyncio.get_event_loop()

    _ = _thread.stack_size(8 * 1024)
    _thread.start_new_thread(loop.run_forever, ())

    # Run WDT Task
    asyncio.create_task(run_wdt())

    # Run Binary Pin
    asyncio.create_task(binary_callback())

if __name__ == '__main__':
    print("MAIN")
    main()
```
[image]

closed 2022-01-30 by straga