micropython/micropython #8023: RP2: Add wiznet5K support Raspberry pi pico

irinakim12, 2021-11-23:
#8021 

I can see the message as below. I think I should just wait for other person with write access. Right?
"this branch has conflicts that must be resolved
Only those with write access to this repository can merge pull requests."

robert-hh, 2021-11-23:
You have to resolve the conflicts first, which are caused by your code. Then the PR is ready for review, which may result in change requests.

irinakim12, 2021-11-25:
The cause of the crash is that I modified ports/rp2/mpconfigport.h .
As far as I know, to use Wiznet5K's library under Network, I need to edit the file.
This error occurs because I do not have permission to edit the file.

irinakim12, 2022-01-19:
@dpgeorge   How can I solve this? Please tell me in detail. WIZnet will have Design contest and we will also use the micropython language at this time.

irinakim12, 2022-04-11:
#8021  
The cause of the crash is that I modified ports/rp2/mpconfigport.h .
As far as I know, to use Wiznet5K's library under Network, I need to edit the file.
This error occurs because I do not have permission to edit the file.

robert-hh, 2022-04-11:
if ports/rp2/mpconfigport.h and the other filed of the rp2 port are local files on your computer, you should have all permissions for changing them, technically.

dpgeorge, 2022-06-03:
Wiznet is supported on rp2 since commit bca816f5ac36609212ab087789ee028d1d5d534d

closed 2022-06-03 by dpgeorge