micropython/micropython #8025: rp2: [Arduino Nano RP2040 Connect] REPL via Serial not working.

Cellebyte, 2021-11-23:
I tried flashing,

* [Lates nightly](https://micropython.org/download/ARDUINO_NANO_RP2040_CONNECT/)

the serial console was not appearing, so I flashed the blink example from here
* [blink.uf2](https://content.arduino.cc/assets/Blink.ino.elf.uf2)
Just to make sure that the flashing still works.

I wiped the memory with the following tool and reflashed the latest nightly from above.
* [flash_nuke.uf2](https://cdn-learn.adafruit.com/assets/assets/000/099/419/original/flash_nuke.uf2?1613329170)

Still not appearing REPL via serial.

```dmesg
[ 2583.404690] usb 1-1: Device not responding to setup address.
[ 2583.611390] usb 1-1: Device not responding to setup address.
[ 2583.817882] usb 1-1: device not accepting address 77, error -71
[ 2583.817996] usb usb1-port1: attempt power cycle
[ 2584.224458] usb 1-1: new full-speed USB device number 78 using xhci_hcd
[ 2584.224599] usb 1-1: Device not responding to setup address.
[ 2584.434692] usb 1-1: Device not responding to setup address.
[ 2584.641228] usb 1-1: device not accepting address 78, error -71
[ 2584.767899] usb 1-1: new full-speed USB device number 79 using xhci_hcd
[ 2584.768298] usb 1-1: Device not responding to setup address.
[ 2584.974668] usb 1-1: Device not responding to setup address.
[ 2585.181168] usb 1-1: device not accepting address 79, error -71
[ 2585.181303] usb usb1-port1: unable to enumerate USB device
```

Cellebyte, 2021-11-23:
The solution I found is exactly the same as for the Feather Board
Solution found here [Solution](https://forum.micropython.org/viewtopic.php?t=10658)

```
#define MICROPY_HW_FLASH_STORAGE_BYTES          (7 * 1024 * 1024 + 256 * 1024)  // 7.25MB flash
#undef PICO_FLASH_SIZE_BYTES
#define PICO_FLASH_SIZE_BYTES                   (8 * 1024 * 1024)
```

Here you can download the current [PATCHED_ARDUINO_NANO_RP2040_CONNECT-20211117-unstable-v1.17-159-ge5f9e2feb.zip](https://github.com/micropython/micropython/files/7591546/PATCHED_ARDUINO_NANO_RP2040_CONNECT-20211117-unstable-v1.17-159-ge5f9e2feb.zip)

dpgeorge, 2021-11-25:
Per the Arduino site, this board is supposed to have 16MiB flash.  You may have a board that has smaller flash installed, for some reason.

@iabdalkader do you know why it would fail?  Does pico-sdk need updating?

Cellebyte, 2021-11-25:
@dpgeorge It fails when I write my project to the chip and then the USB/Serial/REPL is dead.
I have not really debugging capabilities for the board.
Could someone help me to find the issue.

iabdalkader, 2021-11-25:
> @iabdalkader do you know why it would fail? Does pico-sdk need updating?

Yes, newer boards chip with a different flash part (due to chip shortage) which requires a different bootloader. This was fixed and released in pico-sdk 1.3.0 (which in turn requires tinyusb 0.12.0), so they both need to be updated. I already updated both in our fork, I can send a PR.

closed 2021-11-26 by dpgeorge

Cellebyte, 2021-11-27:
@dpgeorge still have issues when I flash a project on to the device.
Directly after the flash micropython seems to work fine.

I tried to flash with pymakr and pico-go. With both my Serial Console went away after board reset. And I can only recover it with flashing micropython  with a different flash size.

iabdalkader, 2021-11-27:
@dpgeorge @Cellebyte There seems to be a different issue with the rp2 port (I managed to reproduce it with RPi Pico and stable release v1.17), I copied a main.py with an LED blink loop to flash and after resetting the board I can't connect anymore to REPL/CDC.

main.py:
```Python
# Blinky example

import time
from machine import Pin

led = Pin(25, Pin.OUT)

while (True):
    led.on()
    time.sleep_ms(250)
    led.off()
    time.sleep_ms(250)
```
To reproduce:
- Flash rp2-pico-20210902-v1.17.uf2 to Pico board.
- Copy main.py (with pico-go or mpremote etc...).
- Reset, should blink LED but ttyACM is non-responsive

> Per the Arduino site, this board is supposed to have 16MiB flash

For some reason, I have configured the board for 8Mbytes (not sure why) but I can fix that later, but it's Not causing any issues.

robert-hh, 2021-11-27:
Is ttyACM non-responsive, but the /dev/ttyACMx device is created at the PC?

iabdalkader, 2021-11-27:
> Is ttyACM non-responsive, but the /dev/ttyACMx device is created at the PC?

Yes exactly, it's like tinyusb task is not called.

robert-hh, 2021-11-27:
If main.py is running a loop, there is no REPL. Can you stop that with Ctrl-C?

Cellebyte, 2021-11-27:
@robert-hh nope not really and there is also no output.

iabdalkader, 2021-11-27:
> If main.py is running a loop, there is no REPL. Can you stop that with Ctrl-C?

Yes that's the issue you can't interrupt main.py loop. With v1.16 you can interrupt it.

Cellebyte, 2021-11-27:
@iabdalkader is it now an issue or my L8 issue ^^
I don't want to create an Issue if there is just me as a dumb enduser doing something wrong :D

robert-hh, 2021-11-27:
The Micropython start message comes once REPL is entered. A print in your main.py however should appear.

Cellebyte, 2021-11-27:
@robert-hh i will test it later the day.

iabdalkader, 2021-11-27:
> I don't want to create an Issue if there is just me as a dumb enduser doing something wrong :D

I don't think so, I think there is an issue. I tested with the older stable release v1.16 (same main.py in flash) and with v1.16 I can connect to ttyACM0 and just Ctrl-C to stop the script and drop to REPL. I can Not do the same with stable release v1.17 (or nightly builds). Note I tested both RPi Pico and Nano, so it's Not specific to the Nano-RP2040. The initial issue you reported with the nano (flashing the firmware bricked it) was fixed .

Cellebyte, 2021-11-27:
@iabdalkader ok so I should create a new one regarding that new issue?

iabdalkader, 2021-11-27:
> . A print in your main.py however should appear.

A print statement in the loop does appear and allows me to interrupt the loop (seems it triggers some tinyusb callback).

> @iabdalkader ok so I should create a new one regarding that new issue?

If you like, this is an unrelated issue, I know for sure your original issue was fixed with pico-sdk + tinyusb updates.

Cellebyte, 2021-11-27:
@iabdalkader in my case the `/dev/ttyACMx` gets completely removed from my linux system and I am not able to connect again to the Boards serial. The error message from above is still showing up when I do a `dmesg -w` on my machine.

I can only get the serial device `/dev/ttyACMx` back when I reflash micropython with a different storage size.

iabdalkader, 2021-11-27:
> My `main.py` looks like this.

Try the blinky main.py, it will let you know if the board is still alive.

Cellebyte, 2021-11-27:
@iabdalkader ok I will try later the day. :+1:

iabdalkader, 2021-11-28:
#7996 Seems related.

commit 97a7cc24 (2021-11-26): lib: Update pico-sdk to 1.3.0 and tinyusb to 0.12.0.

Fixes #8025